indilib/indi #775: SBIG CCD will not connect in v1.7.5 + Raspbian

basilr, 2018-12-08:
Trying to connect 2 SBIG CCD's (ST8E + STL1001) both with same result (see attacked debug log). INDI doesn't see the device on USB1.

Tried from a clean install of Raspban and INDI only, nothing else running on the Pi.
[log_19-15-23.txt](https://github.com/indilib/indi/files/2659328/log_19-15-23.txt)

Both cameras work in INDI v1.7.4.

Feels like a USB driver issue??

Many thanks!

knro, 2018-12-08:
I helped a friend connect his STL cameras just two days ago with the nightly drivers, so I doubt this is the issue.

basilr, 2018-12-08:
Interesting. Was your friend using a Raspberry Pi? Could my problem be a Raspbian issue when I compile from source?

Also, when I power up the CCCD's the no lights come on and the fan doesn't start. It's like I don't have the USB cable even plugged in.

Any thoughts on how I could further troubleshoot my problem? Is there another log file I can look at?  I've tried different Pi boxes, all dependencies are installed (I hope), all packages are up to date, I know the USB cables are good...

I'd really like to use some of the new features in v1.7.5, esp. for Synscan + Celestron mounts.

Many thanks!

basilr, 2018-12-11:
So I think I'm good to go... When compiling from source apparently one needs to install all the Raspian dependencies also, not just the ones needed for the source code. A little awkward because the dependency lists are in 2 different places, github.com and indilib.org.

After installing the Raspian dependencies listed on indilib.org and a reboot, the CCD's fan + lights immediately came on and I was able to connect.

Maybe it would be helpful for others using Raspberry Pi's to mention this on the github site?

Anyway, I still need to test everything on a clear night, but I feel pretty good it will all work.

Many Thanks for all the hard work you and the team do!!

closed 2018-12-11 by basilr