indilib/indi #776: synscan does not know where telescope is pointing (incorrect coordinates)

IvanBenak, 2018-12-10:
The synscan telescope driver, synscandriver.cpp has a few bugs in it regarding retrieving telescope's current RA/DEC position:

sscanf(res,"%lx,%lx#", &n1, &n2)   :  Does not work

as well as:   

ln_get_equ_prec2(&epochPos, JD2000, ln_get_julian_from_sys(), &epochPos);  :  It has the same from and to date structures.

I have attached a diff report indicating the changes I have done to fix the problems.

Thanks,

Ivan.

[synscandriver.diff.txt](https://github.com/indilib/indi/files/2661285/synscandriver.diff.txt)

knro, 2018-12-10:
Thanks, the 2nd issue is already resolved on Git.

I also tested with AZ-GTI using Synscan driver and it worked fine, what's the problem exactly with sscanf(res,"%lx,%lx#", &n1, &n2) : ?

IvanBenak, 2018-12-11:
Pardon me while I chew on some crow as I could have sworn yesterday I couldn't get anything out of the sscanf other than zeros.

Just one thing left then, I hadn't mentioned it specifically as the code update included it.  When you select an object on kstars or on the hand controller where the DEC is negative, it does not report the correct position.  I put this in to fix it:

.
.
    ra  = static_cast<double>(n1) / 0x100000000 * 24.0;
    dec = static_cast<double>(n2) / 0x100000000 * 360.0;
    /* Ivan Benak 2018-12-10 Start */
    if (dec >=180) 
    { 
      dec=dec-360;
    }
    /* Ivan Benak 2018-12-10 End */

After doing that, the negative dec on the hand controller and kstars is properly recognized.

Thanks,

Ivan.

IvanBenak, 2018-12-15:
And based upon me reviewing the description on the update in November, plus the change of the actual line, it looks like you have corrected the DEC issue already as well.  I'd consider this closed then.

Thank you, as everything I have used on it works wonderfully!!

closed 2018-12-15 by IvanBenak