indilib/indi #787: Year in GPSD and INDI client differ

elbarbudo, 2018-12-20:
I installed indi on a Raspry PI 2.
When I connect my old Garmin ETREX gps, I get repeated errors about Date is negative !
Running xgps show 1963 as the year.
But in INDI client connected to GPSD, I have the correct 2018 year.

On different forums about the Garmin GPS I found that the year was often wrong, but that does not explain why xgps and the indi-gpsd plugin gieve different answer.

Out of topic, but how can I tell GPSD to ignore the year send by the gps and use the system year  which is correct

elbarbudo, 2018-12-20:
[image]
[image]
Screen captures showing the issue

elbarbudo, 2019-01-14:
I found that my Garmin Etrex NEVER give the date to gpsd

closed 2019-01-14 by elbarbudo

jochym, 2019-01-16:
The time is _always_ sourced from system time. See: https://github.com/indilib/indi/issues/823