indilib/indi #798: Enforce counter-weight up in eqmod

nadvornik, 2019-01-04:
Hi,

Is anybody using the "Enforce counter-weight up" property as implemented in #550 ?
I have found it is rather unstable and hard to use.
For example:
- the object is 1 sec east of meridian
- the client wants to avoid extra meridian flip so it requests goto with "Enforce counter-weight up" enabled
- the message is delayed by network latency, the object passes meridian meanwhile
- eqmod performs goto to the object which is now west of meridian, with "Enforce counter-weight up" enabled
- it results in the extra meridian flip that should have been avoided

Also, I have another use case, not mentioned in #550: I have clear view only to the west, so I can keep things simple and avoid meridian flips at all.

I'd suggest this:

Add new property "Target Pier Side" with values 
- "Auto" - current behavior, with counterweight down
- "West", "East". - enforce given pier side, abort goto if the target is outside of safe range

maybe also rename  "Enforce counter-weight up" to  "Allow counter-weight up" and use it to extend the safe range 1 hour across meridian. Alternatively this can be given as numeric property.

I can implement it if this is an acceptable change.

nadvornik, 2019-01-06:
ok, see
https://indilib.org/forum/mounts/4383-enforce-counter-weight-up-in-eqmod.html

TallFurryMan, 2019-01-08:
Also check #612, which attempts to implement the safe range part.

closed 2019-02-26 by knro