indilib/indi #800: pmc8 needs method to change mount types

tolson2000, 2019-01-05:


knro, 2019-01-05:
is this just a matter of alias or are there control protocol differences?

tolson2000, 2019-01-05:
Looks like it is gear ratios, microstep differences, steps per revolution type things.

tolson2000, 2019-01-05:
I don't much about INDI coding standards and the like. I found a table in their documentation which shows the different scales for the axis. So do you think all we need to change is the AXIS SCALEs...
#define PMC8_AXIS0_SCALE 4608000.0
#define PMC8_AXIS1_SCALE 4608000.0
to
#define PMC8_AXIS0_SCALE 4147200.0
#define PMC8_AXIS1_SCALE 4147200.0

[image]

I found this information in ES programmers reference...

knro, 2019-01-05:
@mikefulb Hello Mike! What do you think? Maybe a property to select the mount type and then set these accordingly?

mikefulb, 2019-01-14:
> https://github.com/tolson2000/indi
>
> There is still some general cleanup that can be done, but everybody
> with EXOS2 and iEXOS100 are disapointed. This fix at least gets them
> started with correct movements. Thanks.

That looks reasonable.  I was going to make some changes but really 
wanted to get a mount from Jerry so I could test them but seems like 
you've got it covered thanks.

Michael Fulbright

tolson2000, 2019-01-15:
I watched video on doing pull request. Did it. But I only see 3 of the 4 of my changed files in the commit request. The pmc8.h file did not show up. But there are some addtitions to it not in the indi master.

closed 2019-01-28 by tolson2000