indilib/indi #810: Can't save connection settings in Vixen Starbook

not7cd, 2019-01-09:
In #801 I have introduced bug where every time you start `indi_starbook_telescope` you will default to initial IP.

I was unable to fully extend `Connection::Interface`, and I need some help. I believe it's problem in upstream or in my inexperience with pointers and inheritance. Although I wrote my custom `Connection::Curl` like `Connection::TCP` I'm unable to compile because of problems listed below.

```
/home/not7cd/Workspace/indi/3rdparty/indi-starbook/connectioncurl.cpp: In member function ‘virtual bool Connection::Curl::Connect()’:
/home/not7cd/Workspace/indi/3rdparty/indi-starbook/connectioncurl.cpp:51:56: error: ‘virtual bool INDI::DefaultDevice::saveConfig(bool, const char*)’ is protected within this context
             m_Device->saveConfig(true, "DEVICE_ADDRESS");
                                                        ^
In file included from /usr/include/libindi/inditelescope.h:21:0,
                 from /home/not7cd/Workspace/indi/3rdparty/indi-starbook/connectioncurl.h:11,
                 from /home/not7cd/Workspace/indi/3rdparty/indi-starbook/connectioncurl.cpp:5:
/usr/include/libindi/defaultdevice.h:332:18: note: declared protected here
     virtual bool saveConfig(bool silent = false, const char *property = nullptr);
                  ^~~~~~~~~~
/home/not7cd/Workspace/indi/3rdparty/indi-starbook/connectioncurl.cpp: In member function ‘virtual void Connection::Curl::Activated()’:
/home/not7cd/Workspace/indi/3rdparty/indi-starbook/connectioncurl.cpp:72:52: error: ‘virtual bool INDI::DefaultDevice::loadConfig(bool, const char*)’ is protected within this context
         m_Device->loadConfig(true, "DEVICE_ADDRESS");
                                                    ^
In file included from /usr/include/libindi/inditelescope.h:21:0,
                 from /home/not7cd/Workspace/indi/3rdparty/indi-starbook/connectioncurl.h:11,
                 from /home/not7cd/Workspace/indi/3rdparty/indi-starbook/connectioncurl.cpp:5:
/usr/include/libindi/defaultdevice.h:323:18: note: declared protected here
     virtual bool loadConfig(bool silent = false, const char *property = nullptr);
                  ^~~~~~~~~~
CMakeFiles/indi_starbook_telescope.dir/build.make:88: recipe for target 'CMakeFiles/indi_starbook_telescope.dir/connectioncurl.cpp.o' failed
```

knro, 2019-01-09:
Make `Connection::Curl` a child of `Connection::TCP`, would that work fine?

not7cd, 2019-01-09:
No, it did't work. I can't give you any logical explanation why `Connection::TCP` even compiles when `INDI::DefaultDevice` with these protected methods.

not7cd, 2019-01-09:
This is quite hard one to solve. It would need making some protected methods public or will require creation of some sort of a factory (or not). Why this approach? What makes `Connection::` special?

knro, 2019-01-10:
I looked at your `Connection::Curl` implementation and it does not need to access any protected methods directly. You can override whatever function in `Connection::TCP` to do what you want. For example.

```
Connection::Curl::saveConfigItems(FILE*fp)
{
   Connection::TCP::saveConfigItem(fp);

   // Do your own savings here
}
```

not7cd, 2019-01-11:
Can I do it without extending `Connection::TCP`? I don't share anything meaningful with this class and creates unnecessary dependence.

knro, 2019-01-11:
You can submit a generic Curl connection plugin for inclusion in INDI, if that's possible.

not7cd, 2019-01-12:
This will add cURL dependence. Although this class could be used in `libapogee` or in weather info drivers, I'm don't feel that this class is polished enough.

I lost enough time looking for `friend` declarations. This will come up every time when anyone will want to extend `Connection::Interface`.

knro, 2019-01-12:
Sorry about that. I will update the documentation to put a note regarding the friend declarations. Let me know once you have a generic Curl connection plugin that could perhaps be reused elsewhere (e.g. some of the weather drivers).

not7cd, 2019-01-16:
After consultation with other devs I think this is anti-pattern. I will prepare a PR removing friend unless you have good explanation for this solution.
There is no way to inherit friendship. You could make `Connection::Interface` a friend, but then you need method that will access `DefaultDevice`. But you won't be able to overide this method in child class. So again we are on square one.

knro, 2019-01-17:
I'm open to a different solution for the connection interface issue. How can the plugins then get access to the protected and private members of the DefaultDevice after removing Friend? Maybe there is a better design pattern that betters handles this. We cannot simply make all the required methods public, it has to be a different solution.

commit c6c8f0fb (2019-02-02): Enable config saving for DEVICE_ADDRESS

It's a suboptimal solution. It demands to keep track what properties are
declared in Connection::Curl, and then they are saved and loaded
outside: in StarbookDriver class.
Resolve #810
closed 2019-02-02 by knro