indilib/indi #820: indi_lx200stargo: Reduce maximum slew speed when powered with <15V

NicDoebelin, 2019-01-13:
When the Avalon Linear is powered with 12V (7A), the stepper motors stall during long slews. Even when the mount is only loaded with 10kg and well balanced.

In ASCOM they disable the fastest slew speed "ultra" for voltages <15V for the same reason. From the StarGo user manual rev 3.0.1 page 9:

"“Slew”: N. 4 speeds ranges (Low, Medium, Fast, Ultra) depending on the mount “gear ratio” and, for the maximum speed “Ultra”, on the power supply (requires at least 15 VDC)."

Please add an option to reduce the slew speed, so the Avalon mounts can be powered by a 12V battery in the field.

knro, 2019-02-26:
@ken-self Any ideas?

sterne-jaeger, 2019-05-06:
There is an update on my own indilib fork available: https://github.com/sterne-jaeger/indi/tree/stargo-1.5

I will test it a couple of days and will create a pull request.

sterne-jaeger, 2019-05-07:
Closed with PR#945