indilib/indi #821: indi_lx200stargo: Option to disable keypad

NicDoebelin, 2019-01-13:
When the StarGo keypad is disconnected, the mount randomly aborts slews.

How to reproduce:
1. connect the keypad
2. unpark and perform some slews with the keypad
3. disconnect the keypad
4. perform several long slews in EKOS, check if the target object is reached

The mount randomly aborts the slews. This also happens on Windows when the keypad is not connected but not explicitly disabled in the StarGo ASCOM driver. To fix it I have to boot Windows and disable the keypad in the ASCOM driver.

Note that parking is not affected, only long slews.

sterne-jaeger, 2019-04-27:
Nic, I haven't forgotten it, it's on my list as one of the next features.

sterne-jaeger, 2019-05-06:
There is an update on my own indilib fork available: https://github.com/sterne-jaeger/indi/tree/stargo-1.5

I will test it a couple of days and will create a pull request.

sterne-jaeger, 2019-05-07:
Closed with PR#945.

closed 2019-09-16 by knro