indilib/indi #823: Indi GPSUSB module displays the system date, not the GPS date

elbarbudo, 2019-01-14:
I have a Raspberry PI 2 without internet connection.
I bought a gps module connected by USB to it.
On startup, the Raspberry use the time of shutdown to set the clock.
When I connect my GPS to the Raspberry, the xgps command gives me the correct time, but the indi client USBGPS gives me the system time, as if it was the GPS one
[image]
Attached is a capture of the xgps window showing the gps date and time and the indi client window showing the system time instead of the GPS time.

jochym, 2019-01-16:
There is method in this madness - it is by design. The gpsd module uses system time not the gpsd provided time. GPS time is quite unreliable as a timing source - that is why we use system time controlled by ntpd as time source. Using gpsd time directly will remove all the sophistication of ntp from time handling which is sub-optimal. Just make sure tho have good system time and gps module will follow. 
This is of course a compromise but in my opinion this is the best way to do it. Nevertheless, proposed improvements are welcome.

jochym, 2019-01-17:
That is a good suggestion @elbarbudo . And yours is quite a common use case. GPS is a proper solution here but it does not guarentee _continous_ delivery of time. At most gps ca provide "calibrated time points" to the system. It is quite common for the GPS to have prolonged drop-outs of the time signal. That is why it is better to use ntp+gps to _discipline_ system clock and use the system time in the applications. I will change the label in the GUI - the present one is indeed misleading.

jochym, 2019-01-17:
One more thing @elbarbudo : Which driver do you actually use? There is no gpsusb driver in indi. There is indi-gpsd and indi-gpsnmea . They operate in slightly different mode. I assumed we are talking about gpsd driver which I have written. If it is about NMEA driver it is a different case and the author (Jasem I believe) should comment on this.

knro, 2019-09-16:
This is fixed in master.

closed 2019-09-16 by knro