indilib/indi #837: QHY IMG2Pro temperature issue

sgoudelis, 2019-01-27:
I am experiencing an issue with Ekos maintaining and also showing the current temperature when I use it with my IMG2Pro. When Ekos connects to all devices, the temperature of the CCD id visibly update every couple of seconds. When I click on the check button to set the temperature, the TEC leds lights up, but the temperature stops getting updated in the CCD dialog. The temperature does indeed drop but seems to overshoot the user-defined value. If I press the check button again (to set another temperature value) the actually temperature value of the CCD actually shows up on the CCD dialog. Again, it never updates again, only when I press the check button again. Is this an known issue or is it perhaps something on my side ? Anyone knows if the logic that keeps the temperature at a certain level, part of the camera or INDI or Ekos ?

sgoudelis, 2019-01-28:
I just tested the camera with Maxim DL version 6.0.3. The target temperature is reached and held as normal. I would be willing to ship the camera to whoever wants to test out and perhaps fix the issue. Please let me know if that is something of interest.

knro, 2019-02-26:
Please double check with the latest QHY SDK that was released a week ago.

sgoudelis, 2019-02-28:
Thank you for replying. Can you please point out where I can find the SDK? The QHY site is not very friendly and I see that this project here https://github.com/qhyccd-lzr/QHYCCD_Linux is not updated recently.

knro, 2019-02-28:
It's already released with INDI v1.7.6

knro, 2019-09-16:
Please reopen and attach QHY debug logs if you still have the same with the latest SDK.

closed 2019-09-16 by knro