indilib/indi #854: Error when building indi-altaircam

PawelPleskaczynski, 2019-02-09:
Hello, I'm trying to build `indi-altaircam` after building and installing libaltaircam, but I'm getting the following error:
```
indi/build/indi-altaircam on master took 3s 
➜ make
Scanning dependencies of target indi_altair_ccd
[ 50%] Building CXX object CMakeFiles/indi_altair_ccd.dir/indi_altair.cpp.o
In file included from /usr/include/c++/8.2.1/x86_64-pc-linux-gnu/bits/os_defines.h:39,
                 from /usr/include/c++/8.2.1/x86_64-pc-linux-gnu/bits/c++config.h:508,
                 from /usr/include/c++/8.2.1/bits/stl_algobase.h:59,
                 from /usr/include/c++/8.2.1/bits/stl_tree.h:63,
                 from /usr/include/c++/8.2.1/map:60,
                 from /home/pawel/Desktop/indi/3rdparty/indi-altaircam/indi_altair.h:24,
                 from /home/pawel/Desktop/indi/3rdparty/indi-altaircam/indi_altair.cpp:22:
/usr/include/features.h:381:4: warning: #warning _FORTIFY_SOURCE requires compiling with optimization (-O) [-Wcpp]
 #  warning _FORTIFY_SOURCE requires compiling with optimization (-O)
    ^~~~~~~
/home/pawel/Desktop/indi/3rdparty/indi-altaircam/indi_altair.cpp: In member function ‘void ALTAIRCAM::eventPullCallBack(unsigned int)’:
/home/pawel/Desktop/indi/3rdparty/indi-altaircam/indi_altair.cpp:2025:22: error: ‘unique_lock’ is not a member of ‘std’
                 std::unique_lock<std::mutex> guard(ccdBufferLock);
                      ^~~~~~~~~~~
/home/pawel/Desktop/indi/3rdparty/indi-altaircam/indi_altair.cpp:2025:22: note: ‘std::unique_lock’ is defined in header ‘<mutex>’; did you forget to ‘#include <mutex>’?
/home/pawel/Desktop/indi/3rdparty/indi-altaircam/indi_altair.cpp:30:1:
+#include <mutex>
 
/home/pawel/Desktop/indi/3rdparty/indi-altaircam/indi_altair.cpp:2025:22:
                 std::unique_lock<std::mutex> guard(ccdBufferLock);
                      ^~~~~~~~~~~
/home/pawel/Desktop/indi/3rdparty/indi-altaircam/indi_altair.cpp:2025:39: error: ‘mutex’ is not a member of ‘std’
                 std::unique_lock<std::mutex> guard(ccdBufferLock);
                                       ^~~~~
/home/pawel/Desktop/indi/3rdparty/indi-altaircam/indi_altair.cpp:2025:39: note: ‘std::mutex’ is defined in header ‘<mutex>’; did you forget to ‘#include <mutex>’?
/home/pawel/Desktop/indi/3rdparty/indi-altaircam/indi_altair.cpp:2025:52: error: ‘ccdBufferLock’ was not declared in this scope
                 std::unique_lock<std::mutex> guard(ccdBufferLock);
                                                    ^~~~~~~~~~~~~
/home/pawel/Desktop/indi/3rdparty/indi-altaircam/indi_altair.cpp:2025:46: error: ‘guard’ was not declared in this scope
                 std::unique_lock<std::mutex> guard(ccdBufferLock);
                                              ^~~~~
/home/pawel/Desktop/indi/3rdparty/indi-altaircam/indi_altair.cpp:2040:22: error: ‘unique_lock’ is not a member of ‘std’
                 std::unique_lock<std::mutex> guard(ccdBufferLock);
                      ^~~~~~~~~~~
/home/pawel/Desktop/indi/3rdparty/indi-altaircam/indi_altair.cpp:2040:22: note: ‘std::unique_lock’ is defined in header ‘<mutex>’; did you forget to ‘#include <mutex>’?
/home/pawel/Desktop/indi/3rdparty/indi-altaircam/indi_altair.cpp:2040:39: error: ‘mutex’ is not a member of ‘std’
                 std::unique_lock<std::mutex> guard(ccdBufferLock);
                                       ^~~~~
/home/pawel/Desktop/indi/3rdparty/indi-altaircam/indi_altair.cpp:2040:39: note: ‘std::mutex’ is defined in header ‘<mutex>’; did you forget to ‘#include <mutex>’?
/home/pawel/Desktop/indi/3rdparty/indi-altaircam/indi_altair.cpp:2040:52: error: ‘ccdBufferLock’ was not declared in this scope
                 std::unique_lock<std::mutex> guard(ccdBufferLock);
                                                    ^~~~~~~~~~~~~
/home/pawel/Desktop/indi/3rdparty/indi-altaircam/indi_altair.cpp:2040:46: error: ‘guard’ was not declared in this scope
                 std::unique_lock<std::mutex> guard(ccdBufferLock);
                                              ^~~~~
/home/pawel/Desktop/indi/3rdparty/indi-altaircam/indi_altair.cpp:2054:30: error: ‘unique_lock’ is not a member of ‘std’
                         std::unique_lock<std::mutex> guard(ccdBufferLock);
                              ^~~~~~~~~~~
/home/pawel/Desktop/indi/3rdparty/indi-altaircam/indi_altair.cpp:2054:30: note: ‘std::unique_lock’ is defined in header ‘<mutex>’; did you forget to ‘#include <mutex>’?
/home/pawel/Desktop/indi/3rdparty/indi-altaircam/indi_altair.cpp:2054:47: error: ‘mutex’ is not a member of ‘std’
                         std::unique_lock<std::mutex> guard(ccdBufferLock);
                                               ^~~~~
/home/pawel/Desktop/indi/3rdparty/indi-altaircam/indi_altair.cpp:2054:47: note: ‘std::mutex’ is defined in header ‘<mutex>’; did you forget to ‘#include <mutex>’?
/home/pawel/Desktop/indi/3rdparty/indi-altaircam/indi_altair.cpp:2086:18: error: ‘unique_lock’ is not a member of ‘std’
             std::unique_lock<std::mutex> guard(ccdBufferLock);
                  ^~~~~~~~~~~
/home/pawel/Desktop/indi/3rdparty/indi-altaircam/indi_altair.cpp:2086:18: note: ‘std::unique_lock’ is defined in header ‘<mutex>’; did you forget to ‘#include <mutex>’?
/home/pawel/Desktop/indi/3rdparty/indi-altaircam/indi_altair.cpp:2086:35: error: ‘mutex’ is not a member of ‘std’
             std::unique_lock<std::mutex> guard(ccdBufferLock);
                                   ^~~~~
/home/pawel/Desktop/indi/3rdparty/indi-altaircam/indi_altair.cpp:2086:35: note: ‘std::mutex’ is defined in header ‘<mutex>’; did you forget to ‘#include <mutex>’?
/home/pawel/Desktop/indi/3rdparty/indi-altaircam/indi_altair.cpp:2086:48: error: ‘ccdBufferLock’ was not declared in this scope
             std::unique_lock<std::mutex> guard(ccdBufferLock);
                                                ^~~~~~~~~~~~~
/home/pawel/Desktop/indi/3rdparty/indi-altaircam/indi_altair.cpp:2086:42: error: ‘guard’ was not declared in this scope
             std::unique_lock<std::mutex> guard(ccdBufferLock);
                                          ^~~~~
make[2]: *** [CMakeFiles/indi_altair_ccd.dir/build.make:63: CMakeFiles/indi_altair_ccd.dir/indi_altair.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:73: CMakeFiles/indi_altair_ccd.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
```

knro, 2019-02-09:
What OS/GCC?

PawelPleskaczynski, 2019-02-09:
GCC: 8.2.1
OS: 64-bit Linux 4.20

PawelPleskaczynski, 2019-02-14:
@knro did you find a solution to the problem?

knro, 2019-02-26:
No, it compiles fine here. Need to check on GCC 8 perhaps.

PawelPleskaczynski, 2019-02-28:
@knro I tried to install 3rd_party drivers using modified PKGBUILD from here: https://aur.archlinux.org/cgit/aur.git/tree/PKGBUILD?h=libindi_3rdparty where I bumped version number to v1.7.6 and changed checksum. While making package with `makepkg`, I got this error:
```
Scanning dependencies of target indi_gphoto_ccd
[ 16%] Building CXX object indi-gphoto/CMakeFiles/indi_gphoto_ccd.dir/gphoto_ccd.cpp.o
/home/pawel/Desktop/src/indi-1.7.6/3rdparty/indi-gphoto/gphoto_ccd.cpp: In member function ‘virtual bool GPhotoCCD::initProperties()’:
/home/pawel/Desktop/src/indi-1.7.6/3rdparty/indi-gphoto/gphoto_ccd.cpp:429:15: error: ‘class INDI::StreamManager’ has no member named ‘setStreamingExposureEnabled’; did you mean ‘StreamExposureN’?
     Streamer->setStreamingExposureEnabled(false);
               ^~~~~~~~~~~~~~~~~~~~~~~~~~~
               StreamExposureN
/home/pawel/Desktop/src/indi-1.7.6/3rdparty/indi-gphoto/gphoto_ccd.cpp: In member function ‘void GPhotoCCD::streamLiveView()’:
/home/pawel/Desktop/src/indi-1.7.6/3rdparty/indi-gphoto/gphoto_ccd.cpp:1875:47: error: ‘ccdBufferLock’ was not declared in this scope
         std::unique_lock<std::mutex> ccdguard(ccdBufferLock);
                                               ^~~~~~~~~~~~~
/home/pawel/Desktop/src/indi-1.7.6/3rdparty/indi-gphoto/gphoto_ccd.cpp:1875:47: note: suggested alternative: ‘ccdBuffer’
         std::unique_lock<std::mutex> ccdguard(ccdBufferLock);
                                               ^~~~~~~~~~~~~
                                               ccdBuffer
make[2]: *** [indi-gphoto/CMakeFiles/indi_gphoto_ccd.dir/build.make:63: indi-gphoto/CMakeFiles/indi_gphoto_ccd.dir/gphoto_ccd.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:687: indi-gphoto/CMakeFiles/indi_gphoto_ccd.dir/all] Error 2
make: *** [Makefile:152: all] Error 2
==> ERROR: A failure occurred in build().
    Aborting...
```

Once again I got the error with `ccdBufferLock`, which is the same as when I was building Altair drivers. Am I missing some library or it's not my fault? Should I attach whole building log?

PawelPleskaczynski, 2019-03-04:
I tried again, but with GCC 5.5.0, because I found that on your CI system there's also GCC v.5, but I got the same errors :disappointed:

rmathar, 2019-03-07:
The error looks like the compilation is done while some old header files (incompatible with the class specifications of 1.7.6) are hanging around in the build directory. So my first attempt to solve this would be to rm indi-1.7.6/build, mkdir indi-1.7.6/build and to start the cmake from scratch.

PawelPleskaczynski, 2019-03-07:
@rmathar I always started fresh, also when I changed GCC versions

PawelPleskaczynski, 2019-04-09:
I installed updated version from the AUR and now Altair driver is working properly :smiley:

closed 2019-04-09 by PawelPleskaczynski