indilib/indi #856: Errors compiling indi-asi under Ubuntu

Swich1987, 2019-02-11:
I worked remotely with the "zwo asi1600mm-cool" camera. All was fine for a few months, but last time it disappeared from Kstars (and INDI). It suddenly started to show me the message "No ASI cameras detected. Power on?". But the camera is connected and detected through `lsusb` as
`Bus 001 Device 008: ID 04b4:6572 Cypress Semiconductor Corp. `
I tried to reconnect, use asi290, change the cable, reinstall all, but that didn't help. 
So now I'm trying to compile the driver from source, using instructions from 
https://github.com/indilib/indi/blob/master/3rdparty/indi-asi/README.md
but that gets me errors:
```
/home/user/indi/3rdparty/indi-asi/build
user@ubuntu:~/indi/3rdparty/indi-asi/build$ make
[ 12%] Building CXX object CMakeFiles/indi_asi_ccd.dir/asi_ccd.cpp.o
/home/user/indi/3rdparty/indi-asi/asi_ccd.cpp: In member function ‘int ASICCD::grabImage()’:
/home/user/indi/3rdparty/indi-asi/asi_ccd.cpp:1138:5: error: ‘unique_lock’ is not a member of ‘std’
     std::unique_lock<std::mutex> guard(ccdBufferLock);
     ^
/home/user/indi/3rdparty/indi-asi/asi_ccd.cpp:1138:22: error: ‘mutex’ is not a member of ‘std’
     std::unique_lock<std::mutex> guard(ccdBufferLock);
                      ^
/home/user/indi/3rdparty/indi-asi/asi_ccd.cpp:1138:40: error: ‘ccdBufferLock’ was not declared in this scope
     std::unique_lock<std::mutex> guard(ccdBufferLock);
                                        ^
/home/user/indi/3rdparty/indi-asi/asi_ccd.cpp:1138:53: error: ‘guard’ was not declared in this scope
     std::unique_lock<std::mutex> guard(ccdBufferLock);
                                                     ^
/home/user/indi/3rdparty/indi-asi/asi_ccd.cpp: In member function ‘void ASICCD::streamVideo()’:
/home/user/indi/3rdparty/indi-asi/asi_ccd.cpp:1820:9: error: ‘unique_lock’ is not a member of ‘std’
         std::unique_lock<std::mutex> guard(ccdBufferLock);
         ^
/home/user/indi/3rdparty/indi-asi/asi_ccd.cpp:1820:26: error: ‘mutex’ is not a member of ‘std’
         std::unique_lock<std::mutex> guard(ccdBufferLock);
                          ^
/home/user/indi/3rdparty/indi-asi/asi_ccd.cpp:1820:44: error: ‘ccdBufferLock’ was not declared in this scope
         std::unique_lock<std::mutex> guard(ccdBufferLock);
                                            ^
/home/user/indi/3rdparty/indi-asi/asi_ccd.cpp:1820:57: error: ‘guard’ was not declared in this scope
         std::unique_lock<std::mutex> guard(ccdBufferLock);
                                                         ^
CMakeFiles/indi_asi_ccd.dir/build.make:62: recipe for target 'CMakeFiles/indi_asi_ccd.dir/asi_ccd.cpp.o' failed
make[2]: *** [CMakeFiles/indi_asi_ccd.dir/asi_ccd.cpp.o] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/indi_asi_ccd.dir/all' failed
make[1]: *** [CMakeFiles/indi_asi_ccd.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2
```

Info about my system:
```
Linux ubuntu 4.4.0-135-generic #161-Ubuntu SMP Mon Aug 27 10:45:01 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux
cmake/xenial-updates,now 3.5.1-1ubuntu3 amd64 [installed]
cmake-data/xenial-updates,xenial-updates,now 3.5.1-1ubuntu3 all [installed,automatic]
...
indi-asi/xenial,now 1.0~201902060254~ubuntu16.04.1 amd64 [installed]
...
libindi-data/now 1.7.4~201809231300~ubuntu16.04.1 all [installed,upgradable to: 1.7.6~201902051645~ubuntu16.04.1]
libindi1/now 1.7.4~201809231300~ubuntu16.04.1 amd64 [installed,upgradable to: 1.7.6~201902051645~ubuntu16.04.1]
```

I can provide any other needed information.

knro, 2019-02-11:
Did you compile and install libindi first?

Swich1987, 2019-02-11:
Thank you for the fast answer.
I tried different ways, so I don't remember. I'm sure that one time I installed it using `apt-get`. I will try right now to uninstall it using `apt-get remove`, then compile it from the source.

Swich1987, 2019-02-11:
Yes, that resolved the problem with compilation, thank you. Will hope this will help the indi-server to find the camera.

Swich1987, 2019-02-11:
Still having a problem connecting to the camera. Restart indiserver and reboot PC didn't change anything. Is there anything else, what I can try to connect this camera to indiserver? 

> Did you compile and install libindi first?

Swich1987, 2019-02-18:
> Still having a problem connecting to the camera. Restart indiserver and reboot PC didn't change anything. Is there anything else, what I can try to connect this camera to indiserver?
> Maybe I need to open a new issue, or post it on the forum? It compiled successfully.
> 
> > Did you compile and install libindi first?

Maybe I should open it as other issue?

Swich1987, 2019-02-19:
There is no error with compiling, so I'm going to create a new issue for lost connection.

closed 2019-02-19 by Swich1987