indilib/indi #862: Unable to connect ZWO ASI 1600MC-Cool camera to INDI server

Swich1987, 2019-02-19:
I used the ZWO ASI 1600MC-Cool camera as usual for a few months, but suddenly (about a month ago) it lost the connection.
Now every time I try to connect to it show the message **`No ASI cameras detected. Power on?`** But the camera is connected and detected through **`lsusb`** as
```
Bus 001 Device 008: ID 04b4:6572 Cypress Semiconductor Corp.
``` 
I tried to reconnect, connect asi290 camera instead (which is Ok for sure, but also not recognized), change the cable, reinstall everything, compile from source, restart PC, but that didn't help.
I compiled the driver from source, using instructions from
github.com/indilib/indi/blob/master/3rdparty/indi-asi/README.md 
that also didn't help to resolve the problem. I tried to compile a different version of indi_asi_ccd driver (1.7.5, 1.7.4, 1.7.2, 1.7.1, 1.7.0, 1.5.0), that also gave nothing.

The camera is connected to this PC (only command line, so I can't check it using other apps):
```
Description:    Ubuntu 16.04.5 LTS
Release:        16.04
Codename:       xenial
```

Log from indiserver:
```
user@ubuntu:~/indi/3rdparty/indi-asi/build$ indiserver -vvv indi_asi_ccd
2019-02-18T12:13:07: startup: indiserver -vvv indi_asi_ccd
2019-02-18T12:13:07: Driver indi_asi_ccd: pid=19740 rfd=3 wfd=6 efd=7
2019-02-18T12:13:07: listening to port 7624 on fd 4
2019-02-18T12:13:07: Driver indi_asi_ccd: sending msg copy 1 nq 1:
<getProperties version='1.7'/>

2019-02-18T12:13:07: Driver indi_asi_ccd: read message    'No ASI cameras detected. Power on?'
2019-02-18T12:13:12: Client 0: new arrival from 95.71.111.111:51796 - welcome!
2019-02-18T12:13:12: Client 0: read getProperties
2019-02-18T12:13:12: Driver indi_asi_ccd: queuing responsible for <getProperties device='' name=''>
2019-02-18T12:13:12: Driver indi_asi_ccd: sending msg copy 1 nq 1:
<getProperties version="1.7"/>

2019-02-18T12:13:12: Driver indi_asi_ccd: read message    'No ASI cameras detected. Power on?'
2019-02-18T12:13:12: Client 0: queuing <message device='' name=''>
2019-02-18T12:13:12: Client 0: sending msg copy 1 nq 1:
<message timestamp="2019-02-18T12:13:12" message="No ASI cameras detected. Power on?"/>

2019-02-18T12:13:15: Client 0: read EOF
2019-02-18T12:13:15: Client 0: shut down complete - bye!
```

knro, 2019-02-19:
Did you also install asi-common which includes the 99-asi.rules file? It needs to be placed in /lib/udev/rules.d/ directory

Swich1987, 2019-02-20:
> Did you also install asi-common which includes the 99-asi.rules file? It needs to be placed in /lib/udev/rules.d/ directory

No, I didn't.
I made the procedure described [here](https://astronomy-imaging-camera.com/manuals/ASI%20Cameras%20software%20Manual%20Linux%20OSX%20EN.pdf) at page 4 under "1. Install udev rule(Linux)" title. So I create a file **/lib/udev/rules.d/asi.rules** in which I wrote next:
```
ACTION=="add", ATTR{idVendor}=="04b4", RUN+="/bin/sh -c '/bin/echo 200 >/sys/module/usbcore/parameters/usbfs_memory_mb'"
# All ASI Cameras and filter wheels
SUBSYSTEMS=="usb", ATTR{idVendor}=="04b4", MODE="0777"

```

Let me look how to install asi-common, install it, and and I will give you the answer as soon as possible.

Swich1987, 2019-02-20:
Thank you for your help, looks like it was a connection issue because of bad contact on USB port and mistake in our team. When we checked if it's connected by **`lsusb`**, it was wrong device. The correct camera vendor is  **03c3:1601**, not **04b4:6572**. So the camera was not connected.

closed 2019-02-20 by Swich1987