indilib/indi #863: Unable to connect Lacerta MFOC

joerggrosse-kracht, 2019-02-20:
I can't connect to Lacerta MFOC device. 
I got a error message "device or program in use of another process"
The original Code from the PR works for me.

knro, 2019-02-20:
Perhaps this has less to do with the driver and more to do with the changes introduced to indicom regarding tty_connect. Do you recompile libindi and installed it on the same?

joerggrosse-kracht, 2019-02-20:
yes i have recompile libindi and installed on the same host Pi 3b+ with mate 16.04 (stellarmate),
My other devices (also new recompiled) Metostation (indi-duino), EQmod connect without problems.

knro, 2019-02-20:
The exact error is like this "Port /dev/ttyUSB0 is already used by another driver or process" ? How many ports are detected in the Connections tab of the driver?

joerggrosse-kracht, 2019-02-21:
The exact message was "2019-02-21T13:07:20:[WARNING] Port /dev/focuser is already used by another driver or process
System Ports in the connection tab are /dev/ttyUSB0, /dev/ttyUSB1 and /dev/ttyUSB2
/dev/ttyUSB0 (Lacerta MFOC) is mapped to /dev/focuser i have checked the symlink created by udev under /dev

knro, 2019-02-21:
I just pushed a fix that can help in this.. can you git pull and try again?

joerggrosse-kracht, 2019-02-21:
same issue, i have tested now with only the MFOC connected to USB and Auto Search
with following result:

[WARNING] Port /dev/ttyUSB3 is already used by another driver or Process.
[INFO] Trying connecting to /dev/ttyUSB3 @ 9600
[WARNING] Communication with /dev/ttyUSB0 @ 9600 failed. Starting Auto Search

knro, 2019-02-21:
If it is the only thing connected, how come it found /dev/ttyUSB3??

joerggrosse-kracht, 2019-02-21:
I have all devices removed, and after plugin the MFOC it appears as /dev/ttyUSB3

joerggrosse-kracht, 2019-02-21:
If i reboot the PI only with MFOC the Port ist /dev/ttyUSB0

 

But the problem is always the same Port is already used

knro, 2019-02-21:
Ok the driver was missing a handshake function. It is added now, please git pull and test.

joerggrosse-kracht, 2019-02-22:
Great Job !
the connect works now fine.

closed 2019-02-26 by knro