indilib/indi #870: dev version of Ekos/Focuser Tab/Crash on double click on the captured image

Patrick-81, 2019-02-26:
I update on one of my PC the dev version (bleeding edge) of Kstars/Ekos/Indi.
I have some issues with this version. Which is normal but need to be mentionned.
The issue: when doing manual focusing if I double clic in the captured image to move the green box on a star I get a crash of Kstars/Ekos. Here joined a log of a quick daily session but the behavior is the same with stars in the field.
[log_14-05-53.txt](https://github.com/indilib/indi/files/2905645/log_14-05-53.txt)
[indi_nikon_ccd_13_23_39.log](https://github.com/indilib/indi/files/2905647/indi_nikon_ccd_13_23_39.log)

knro, 2019-02-26:
This doesn't show the crash. You can run KStars via gdb:
`gdb -ex run --args kstars`

Then if it crashes, type **bt** to get the backtrace and paste it here.

Patrick-81, 2019-02-26:
I tried it but it doesn't crashed in debug mode.

knro, 2019-02-26:
Are you compiling KStars yourself? if you type in `which kstars `
what do you get?

Patrick-81, 2019-02-26:
/usr/bin/kstars

knro, 2019-02-26:
well, not sure what to try. Keep running KStars in gdb until you have a crash and then report back.

Patrick-81, 2019-02-26:
Ok thank you Jasem

Patrick-81, 2019-02-26:
Finally after a reinstallation

1rst try

Thread 1 "kstars" received signal SIGSEGV, Segmentation fault.
__memcpy_avx_unaligned ()
    at ../sysdeps/x86_64/multiarch/memcpy-avx-unaligned.S:51
51	../sysdeps/x86_64/multiarch/memcpy-avx-unaligned.S: Aucun fichier ou dossier de ce type.
(gdb) tb
Temporary breakpoint 1 at 0x7ffff1881ec2: file ../sysdeps/x86_64/multiarch/memcpy-avx-unaligned.S, line 51.
(gdb) 

2nd try to confirm

Thread 1 "kstars" received signal SIGSEGV, Segmentation fault.
0x00005555558e116f in FITSLabel::mouseMoveEvent (this=0x5555597508f0, 
    e=0x7fffffffcab0)
    at /build/kstars-bleeding-IKllhF/kstars-bleeding-3.1.0+201902260427~ubuntu16.04.1/kstars/fitsviewer/fitslabel.cpp:123
123	/build/kstars-bleeding-IKllhF/kstars-bleeding-3.1.0+201902260427~ubuntu16.04.1/kstars/fitsviewer/fitslabel.cpp: Aucun fichier ou dossier de ce type.
(gdb) tb
Temporary breakpoint 1 at 0x5555558e116f: file /build/kstars-bleeding-IKllhF/kstars-bleeding-3.1.0+201902260427~ubuntu16.04.1/kstars/fitsviewer/fitslabel.cpp, line 123.
(gdb)

knro, 2019-02-26:
in gdb, when it crashes try to print the values. What's the size of your image?

```
print width
print x
print y

```

Patrick-81, 2019-02-27:
Another crash while registering darks. I tried to consult the stats on the viewer.
Thread 1 "kstars" received signal SIGSEGV, Segmentation fault.
0x00007fffe4650b1a in QFontEngineFT::loadGlyphSet(QTransform const&) ()
   from /usr/lib/x86_64-linux-gnu/libQt5XcbQpa.so.5
(gdb) tb
Temporary breakpoint 1 at 0x7fffe4650b1a

knro, 2019-02-27:
it's **bt** or **backtrace**

Patrick-81, 2019-02-27:
sorry made a confusion between backtrace(bt) and tarceback(tb)
Here are the results of 2 crashes:
1/ doing dark files and asking for statistics once in the fitsviewer and asking again:
Thread 1 "kstars" received signal SIGSEGV, Segmentation fault.
0x00007fffe4650b1a in QFontEngineFT::loadGlyphSet(QTransform const&) ()
   from /usr/lib/x86_64-linux-gnu/libQt5XcbQpa.so.5
(gdb) bt
#0  0x00007fffe4650b1a in QFontEngineFT::loadGlyphSet(QTransform const&) ()
    at /usr/lib/x86_64-linux-gnu/libQt5XcbQpa.so.5
#1  0x00007fffe4650b98 in QFontEngineFT::loadGlyphFor(unsigned int, QFixed, QFontEngine::GlyphFormat, QTransform const&, bool) ()
    at /usr/lib/x86_64-linux-gnu/libQt5XcbQpa.so.5
#2  0x00007fffe4651c64 in QFontEngineFT::lockedAlphaMapForGlyph(unsigned int, QFixed, QFontEngine::GlyphFormat, QTransform const&, QPoint*) ()
    at /usr/lib/x86_64-linux-gnu/libQt5XcbQpa.so.5
#3  0x00007ffff46730f1 in QRasterPaintEngine::drawCachedGlyphs(int, unsigned int const*, QFixedPoint const*, QFontEngine*) ()
    at /usr/lib/x86_64-linux-gnu/libQt5Gui.so.5
#4  0x00007ffff4675c5f in QRasterPaintEngine::drawTextItem(QPointF const&, QTextItem const&) () at /usr/lib/x86_64-linux-gnu/libQt5Gui.so.5
#5  0x00007ffff46932d1 in  () at /usr/lib/x86_64-linux-gnu/libQt5Gui.so.5
#6  0x00007ffff4520c5d in QTextLine::draw(QPainter*, QPointF const&, QTextLayout::FormatRange const*) const () at /usr/lib/x86_64-linux-gnu/libQt5Gui.so.5
#7  0x00007ffff4521a0d in QTextLayout::draw(QPainter*, QPointF const&, QVector<QTextLayout::FormatRange> const&, QRectF const&) const ()
    at /usr/lib/x86_64-linux-gnu/libQt5Gui.so.5
#8  0x00007ffff4ded52b in QWidgetLineControl::draw(QPainter*, QPoint const&, QRect const&, int) () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#9  0x00007ffff4d61078 in QLineEdit::paintEvent(QPaintEvent*) ()
    at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
---Type <return> to continue, or q <return> to quit---
#10 0x00007ffff4c52fc8 in QWidget::event(QEvent*) ()
    at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#11 0x00007ffff4d66389 in QLineEdit::event(QEvent*) ()
    at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#12 0x00007ffff4c1005c in QApplicationPrivate::notify_helper(QObject*, QEvent*) () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#13 0x00007ffff4c15516 in QApplication::notify(QObject*, QEvent*) ()
    at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#14 0x00007ffff3e2438b in QCoreApplication::notifyInternal(QObject*, QEvent*) () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#15 0x00007ffff4c4bab9 in QWidgetPrivate::sendPaintEvent(QRegion const&) ()
    at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#16 0x00007ffff4c4c101 in QWidgetPrivate::drawWidget(QPaintDevice*, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) ()
    at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#17 0x00007ffff4c4cdac in QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#18 0x00007ffff4c4cc99 in QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#19 0x00007ffff4c4cc99 in QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, int, QPainter*, QWid---Type <return> to continue, or q <return> to quit---
getBackingStore*) () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#20 0x00007ffff4c4cc99 in QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#21 0x00007ffff4c4cc99 in QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#22 0x00007ffff4c4cc99 in QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#23 0x00007ffff4c4cc99 in QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#24 0x00007ffff4c4cc99 in QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#25 0x00007ffff4c4cc99 in QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#26 0x00007ffff4c4cc99 in QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#27 0x00007ffff4c4cc99 in QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, ---Type <return> to continue, or q <return> to quit---
QList<QObject*> const&, int, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
[... cut ...]
#96 0x00007ffff3e7a7cf in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#97 0x00007ffff3e21b4a in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#98 0x00007ffff3e29bec in QCoreApplication::exec() ()
    at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#99 0x0000555555625f00 in main(int, char**) (argc=1, argv=<optimized out>)
    at /build/kstars-bleeding-IKllhF/kstars-bleeding-3.1.0+201902260427~ubuntu16.04.1/kstars/main.cpp:331

Patrick-81, 2019-02-27:
2/ doing focus and clicking twice in the image dispaly

</setNumberVector>
[New Thread 0x7fffba7fd700 (LWP 15285)]
d201: newSwitchVector with no valid members
d201: newSwitchVector with no valid members
d201: newSwitchVector with no valid members
[New Thread 0x7fffb9ffc700 (LWP 15330)]

Thread 1 "kstars" received signal SIGSEGV, Segmentation fault.
__memcpy_avx_unaligned ()
    at ../sysdeps/x86_64/multiarch/memcpy-avx-unaligned.S:51
51	../sysdeps/x86_64/multiarch/memcpy-avx-unaligned.S: Aucun fichier ou dossier de ce type.
(gdb) bt
#0  0x00007ffff1881ec2 in __memcpy_avx_unaligned ()
    at ../sysdeps/x86_64/multiarch/memcpy-avx-unaligned.S:51
#1  0x000055555580cda0 in FITSData::findCannyStar<unsigned short>(FITSData*, QRect const&) (__len=<optimized out>, __src=0x7fff981749f4, __dest=<optimized out>) at /usr/include/x86_64-linux-gnu/bits/string3.h:53
#2  0x000055555580cda0 in FITSData::findCannyStar<unsigned short>(FITSData*, QRect const&) (data=data@entry=0x555559e6e9e0, boundary=...)
    at /build/kstars-bleeding-IKllhF/kstars-bleeding-3.1.0+201902260427~ubuntu16.04.1/kstars/fitsviewer/fitsdata.cpp:1135
#3  0x0000555555811ac3 in FITSData::findCannyStar(FITSData*, QRect const&) (data=data@entry=0x555559e6e9e0, boundary=...)
    at /build/kstars-bleeding-IKllhF/kstars-bleeding-3.1.0+201902260427~ubuntu16.04.1/kstars/fitsviewer/fitsdata.cpp:1037
#4  0x0000555555811c03 in FITSData::findStars(StarAlgorithm, QRect const&) (this=0x555559e6e9e0, algorithm=ALGORITHM_GRADIENT, trackingBox=...)
    at /build/kstars-bleeding-IKllhF/kstars-bleeding-3.1.0+201902260427~ubuntu16.04.1/kstars/fitsviewer/fitsdata.cpp:1070
#5  0x0000555555a06175 in Ekos::Focus::setCaptureComplete() (this=this@entry=0x555559b14430)
    at /build/kstars-bleeding-IKllhF/kstars-bleeding-3.1.0+201902260427~ubuntu16.04.1/kstars/ekos/focus/focus.cpp:1262
#6  0x0000555555a07688 in Ekos::Focus::newFITS(IBLOB*) (this=0x555559b14430, bp=<optimized out>)
---Type <return> to continue, or q <return> to quit---
    at /build/kstars-bleeding-IKllhF/kstars-bleeding-3.1.0+201902260427~ubuntu16.04.1/kstars/ekos/focus/focus.cpp:1189
#7  0x00007ffff3e52baf in QMetaObject::activate(QObject*, int, int, void**) ()
    at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#8  0x0000555555835672 in ISD::GDInterface::BLOBUpdated(IBLOB*) (this=this@entry=0x55555a226db0, _t1=_t1@entry=0x7fffac0183e0)
    at /build/kstars-bleeding-IKllhF/kstars-bleeding-3.1.0+201902260427~ubuntu16.04.1/obj-x86_64-linux-gnu/kstars/moc_indistd.cpp:384
#9  0x00005555558bd280 in QtPrivate::QFunctorSlotObject<ISD::CCD::loadImageInView(IBLOB*, ISD::CCDChip*)::<lambda()>, 0, QtPrivate::List<>, void>::impl(int, QtPrivate::QSlotObjectBase *, QObject *, void **, bool *) (__closure=<optimized out>)
    at /build/kstars-bleeding-IKllhF/kstars-bleeding-3.1.0+201902260427~ubuntu16.04.1/kstars/indi/indiccd.cpp:1585
#10 0x00005555558bd280 in QtPrivate::QFunctorSlotObject<ISD::CCD::loadImageInView(IBLOB*, ISD::CCDChip*)::<lambda()>, 0, QtPrivate::List<>, void>::impl(int, QtPrivate::QSlotObjectBase *, QObject *, void **, bool *) (arg=<optimized out>, f=...) at /usr/include/x86_64-linux-gnu/qt5/QtCore/qobjectdefs_impl.h:495
#11 0x00005555558bd280 in QtPrivate::QFunctorSlotObject<ISD::CCD::loadImageInView(IBLOB*, ISD::CCDChip*)::<lambda()>, 0, QtPrivate::List<>, void>::impl(int, QtPrivate::QSlotObjectBase *, QObject *, void **, bool *) (arg=<optimized out>, f=...) at /usr/include/x86_64-linux-gnu/qt5/QtCore/qobjectdefs_impl.h:552
#12 0x00005555558bd280 in QtPrivate::QFunctorSlotObject<ISD::CCD::loadImageInVie---Type <return> to continue, or q <return> to quit---
w(IBLOB*, ISD::CCDChip*)::<lambda()>, 0, QtPrivate::List<>, void>::impl(int, QtPrivate::QSlotObjectBase *, QObject *, void **, bool *) (which=<optimized out>, this_=<optimized out>, r=<optimized out>, a=<optimized out>, ret=<optimized out>) at /usr/include/x86_64-linux-gnu/qt5/QtCore/qobject_impl.h:192
#13 0x00007ffff3e52baf in QMetaObject::activate(QObject*, int, int, void**) ()
    at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#14 0x00005555557cb501 in FITSView::loadInFrame() (this=0x555559b60f00)
    at /build/kstars-bleeding-IKllhF/kstars-bleeding-3.1.0+201902260427~ubuntu16.04.1/kstars/fitsviewer/fitsview.cpp:423
#15 0x00007ffff3e52baf in QMetaObject::activate(QObject*, int, int, void**) ()
    at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#16 0x00007ffff3c3b0ee in QFutureWatcherBase::event(QEvent*) ()
    at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#17 0x00007ffff4c1005c in QApplicationPrivate::notify_helper(QObject*, QEvent*) () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#18 0x00007ffff4c15516 in QApplication::notify(QObject*, QEvent*) ()
    at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#19 0x00007ffff3e2438b in QCoreApplication::notifyInternal(QObject*, QEvent*) () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#20 0x00007ffff3e26786 in QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*) () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#21 0x00007ffff3e7a3c3 in  () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#22 0x00007fffede98197 in g_main_context_dispatch (context=0x7fffdc0016f0)
---Type <return> to continue, or q <return> to quit---
    at /build/glib2.0-7ZsPUq/glib2.0-2.48.2/./glib/gmain.c:3154
#23 0x00007fffede98197 in g_main_context_dispatch (context=context@entry=0x7fffdc0016f0) at /build/glib2.0-7ZsPUq/glib2.0-2.48.2/./glib/gmain.c:3769
[... cut ...]
#26 0x00007ffff3e7a7cf in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#27 0x00007ffff3e21b4a in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#28 0x00007ffff3e29bec in QCoreApplication::exec() ()
    at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#29 0x0000555555625f00 in main(int, char**) (argc=1, argv=<optimized out>)
    at /build/kstars-bleeding-IKllhF/kstars-bleeding-3.1.0+201902260427~ubuntu16.04.1/kstars/main.cpp:331

Patrick-81, 2019-02-27:
Another issue: when trying to capture biases with a 0.08s (for the d5300) the capture time is really long and there is only one download and no incrementing of the number of the capture. But I need to investigate more this point as the '.' in the name of the capture when asking to include the exposure time in the image name looks to play a role in this issue.

knro, 2019-03-10:
Ok these issues should really be filed over at bugs.kde.org as we can only look after the INDI bugs here.

closed 2019-03-10 by knro