indilib/indi #872: Error in indi-atik compilation

jmichelet, 2019-02-28:
Trying to compile the indi-atik drivers on an i386 environment, I am facing a compilation error :

home/jaq/bin/indi/3rdparty/indi-atik/atik_ccd.cpp:1291:9: error: ‘ArtemisSetDebugCallbackContext’ was not declared in this scope
         ArtemisSetDebugCallbackContext(this, &ATIKCCD::debugCallbackHelper);
         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/jaq/bin/indi/3rdparty/indi-atik/atik_ccd.cpp:1291:9: note: suggested alternative: ‘ArtemisSetDebugCallback’
         ArtemisSetDebugCallbackContext(this, &ATIKCCD::debugCallbackHelper);
         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
         ArtemisSetDebugCallback

/------------------------------------------------------/
The whole compilation traces :

jaq@iparizar:~/bin/indi/build/indi-atik$ cmake -DCMAKE_INSTALL_PREFIX=/usr -DCMAKE_BUILD_TYPE=Debug ../../3rdparty/indi-atik
-- The CXX compiler identification is GNU 7.3.0
-- The C compiler identification is GNU 7.3.0
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Found CFITSIO 3.43: /usr/lib/i386-linux-gnu/libcfitsio.so
-- Found INDI: /usr/lib/i386-linux-gnu/libindidriver.so;/usr/lib/i386-linux-gnu/libindiAlignmentDriver.so (found version "1.7.6") 
-- Found ZLIB: /usr/lib/i386-linux-gnu/libz.so (found version "1.2.11") 
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Found USB1: /usr/lib/i386-linux-gnu/libusb-1.0.so (found version "1.0.21") 
-- Performing Test USB1_HAS_LIBUSB_ERROR_NAME
-- Performing Test USB1_HAS_LIBUSB_ERROR_NAME - Success
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found Atik Library: /usr/lib/i386-linux-gnu/libatikcameras.so
-- Performing Test COMPATIBLE_FORTIFY_SOURCE
-- Performing Test COMPATIBLE_FORTIFY_SOURCE - Success
-- Configuring done
-- Generating done
-- Build files have been written to: /home/jaq/bin/indi/build/indi-atik
jaq@iparizar:~/bin/indi/build/indi-atik$ make
Scanning dependencies of target indi_atik_wheel
[ 25%] Building CXX object CMakeFiles/indi_atik_wheel.dir/atik_wheel.cpp.o
[ 50%] Linking CXX executable indi_atik_wheel
[ 50%] Built target indi_atik_wheel
Scanning dependencies of target indi_atik_ccd
[ 75%] Building CXX object CMakeFiles/indi_atik_ccd.dir/atik_ccd.cpp.o
/home/jaq/bin/indi/3rdparty/indi-atik/atik_ccd.cpp: In member function ‘virtual void ATIKCCD::debugTriggered(bool)’:
/home/jaq/bin/indi/3rdparty/indi-atik/atik_ccd.cpp:1291:9: error: ‘ArtemisSetDebugCallbackContext’ was not declared in this scope
         ArtemisSetDebugCallbackContext(this, &ATIKCCD::debugCallbackHelper);
         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/jaq/bin/indi/3rdparty/indi-atik/atik_ccd.cpp:1291:9: note: suggested alternative: ‘ArtemisSetDebugCallback’
         ArtemisSetDebugCallbackContext(this, &ATIKCCD::debugCallbackHelper);
         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
         ArtemisSetDebugCallback
/home/jaq/bin/indi/3rdparty/indi-atik/atik_ccd.cpp:1293:9: error: ‘ArtemisSetDebugCallbackContext’ was not declared in this scope
         ArtemisSetDebugCallbackContext(nullptr, nullptr);
         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/jaq/bin/indi/3rdparty/indi-atik/atik_ccd.cpp:1293:9: note: suggested alternative: ‘ArtemisSetDebugCallback’
         ArtemisSetDebugCallbackContext(nullptr, nullptr);
         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
         ArtemisSetDebugCallback
CMakeFiles/indi_atik_ccd.dir/build.make:62: recipe for target 'CMakeFiles/indi_atik_ccd.dir/atik_ccd.cpp.o' failed
make[2]: *** [CMakeFiles/indi_atik_ccd.dir/atik_ccd.cpp.o] Error 1
CMakeFiles/Makefile2:104: recipe for target 'CMakeFiles/indi_atik_ccd.dir/all' failed
make[1]: *** [CMakeFiles/indi_atik_ccd.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2

jmichelet, 2019-03-01:
Note that this error does not happen in a 64-bit environment, even with the very same compiler flavour (GNU 7.3.0). Perhaps is the issue related to the 32-bit libatik library ?
[rapport_compil_indi_atik_64b.txt](https://github.com/indilib/indi/files/2918742/rapport_compil_indi_atik_64b.txt)

glowmouse, 2019-03-12:
I looked at the Atik include file in libatik.  The definition for ArtemisSetDebugCallbackContext looks reasonable to my eye.  i.e.,

```
    void ArtemisSetDebugCallback(void(*callback)(const char *message));
  void ArtemisSetDebugCallbackContext(void * context, void(*callback)(void *context, const char *message));
  void ArtemisSetFirmwareDir(const char * firmwareDir);
```

I didn't spot anything that stood out as a possible cause.  Some things that make help, if you haven't already tried them...

I wonder if your i386 CMake is picking up an older Atik Library?  i.e.,  if you rebuild and reinstalll indi/3rdparty/libatik does it over-write the library at /usr/lib/i386-linux-gnu/?

Similarly, I wonder if the include file on your system matches the one in libatik.  i.e., on my system I don't see any differences (as expected).

```
andrew@miles:~/src/indi/3rdparty/indi-atik$ cd /usr
andrew@miles:/usr$ find | grep AtikCameras.h
./local/include/libatik/AtikCameras.h
andrew@miles:/usr$ diff ./local/include/libatik/AtikCameras.h /home/andrew/src/indi/3rdparty/libatik/AtikCameras.h 
andrew@miles:/usr$ 
```

knro, 2019-04-23:
Can you replicate this issue with the latest code in master?

TallFurryMan, 2019-07-11:
For what it's worth, I can't build indi-atik with gcc 4.9 on i386. Minor as I can still use Cloudmakers' RPM.

TallFurryMan, 2019-09-07:
Building `indi-atik` succeeded on my i386 Lubuntu 18.04 system using 0bd6abc1, and the driver is functional. I note I had to build `libatik` separately and execute `ldconfig` manually when installing to `/usr/local` instead of `/usr`.

closed 2019-09-16 by knro