indilib/indi #881: Synscan driver doesn't work on unaligned mounts anymore

artix75, 2019-03-09:
Hello, after updating to 1.7.6 I've noticed that I cannot connect my dobson synscan mount anymore without previously aligning it (INDI prevent connecting to the mount and returns the error "Mount is not aligned. Please align the mount first and connect again.").
Why did you put such a constraint like this?
My mount can also works without star alignment, in tracking mode (GOTO disabled). By this way I cannot remotely control it anymore if I don't perform the star alignment (ie. slew, enable/disable tracking).

The synscan handcontroller give users the opportunity (and freedom) to use it also without alignment, but your driver now requires a mandatory alignment.

I also cannot make tests (ie. with joystick) or even develope my client software in the daytime inside my room, since I'm always forced to perform a star alignment in order to use INDI.

It makes completely no sense to me. I think that a driver should leave the client software free to do whatever it wants with the hardware, without putting these kind of constraints. It's up to the client to warn user about missing alignment or prevent them using it, not to the driver IMHO.

I think you should remove this contraint or, at least, make it optional.

knro, 2019-03-09:
The older driver was renamed to indi_synscanlegacy_telescope, you could use that.

Though for remote control, it might be a better idea with use EQMod with some EQDir cable instead of relying on the handset.

artix75, 2019-03-09:
@knro Ok, I'll try the legacy one. Anyway I still don't understand why the new driver requires star alignment as mandatory. Is it necessary?

Thank you!

knro, 2019-03-10:
It is necessary for the updated driver yes

closed 2019-03-10 by knro

WLR86, 2023-07-02:
Hi 

Just trying to understand... now how would I go about using synscan driver for daylight astronomy ? Also I tried legacy driver (latest available as of now) and it requires alignment too.

On a side note, If I would want to set date and location using INDI before aligning, I can't either because "it's not aligned" - I don't get why this is mandatory - and why this can't be switched on/off depending on the use case ?