indilib/indi #883: v1.7.6 Compile error in CENTOS7 (latest) when compiling indi_pegasus_upb

peter-englmaier, 2019-03-10:
I tried to compile the newest version v1.7.6 with CENTOS7 and get the following error (v1.7.5 compiled fine):

Scanning dependencies of target indi_pegasus_upb
[ 20%] Building CXX object CMakeFiles/indi_pegasus_upb.dir/drivers/auxiliary/pegasus_upb.cpp.o
[ 20%] Linking CXX executable indi_pegasus_upb
CMakeFiles/indi_pegasus_upb.dir/drivers/auxiliary/pegasus_upb.cpp.o: In function `PegasusUPB::split(std::string const&, std::string const&)':
/home/ppe/src/indilib/indi/libindi/drivers/auxiliary/pegasus_upb.cpp:1056: undefined reference to `std::regex_token_iterator<__gnu_cxx::__normal_iterator<char const*, std::string>, char, std::regex_traits<char> >::regex_token_iterator(__gnu_cxx::__normal_iterator<char const*, std::string>, __gnu_cxx::__normal_iterator<char const*, std::string>, std::basic_regex<char, std::regex_traits<char> > const&, int, std::bitset<11ul>)'

... and several more...

I tried installing everything mentioned in the docker file, but same error. 
Any idea what might be wrong in my setup?
Thanks, Peter

rmathar, 2019-03-14:
The regex header (I was trying to include the triangular brackets here but the code conventions of this interface here do not allow to do that) was implemented in gcc in gcc 4.9.0. Check the output of ''gcc --version''. If obsolete get a newer compiler as documented in https://www.softwarecollections.org/en/scls/rhscl/devtoolset-7/

peter-englmaier, 2019-03-14:
Indeed, I have:
gcc (GCC) 4.8.5 20150623 (Red Hat 4.8.5-36)
With devtoolset-7 it works! Excellent! 
I'm closing the issue!

closed 2019-03-14 by peter-englmaier