indilib/indi #887: Please add support for ALccd-QHY 9 monochrome

peter-englmaier, 2019-03-14:
Dear Developer

Pleaes add support for this device, USB ID 1618:8301. I can provide output of qhy_ccd_test or test alpha/beta releases if needed.

Best, Peter

knro, 2019-03-16:
It's already in added in the rules files. What's the issue?

peter-englmaier, 2019-03-16:
Hi Jasem
I use the latest kstars-bleeding, indi-full with depending packages in the latest ubuntu bionic.

With this setup, I got initially only a yellow status in indi driver and nothing more. However, after reinstalling packages and resetting configurations I have got a 'green' light when connecting. 
Still, when I try to do an image I get timeouts:
2019-03-16T13:59:51 Exposure timeout. Restarting exposure...
2019-03-16T13:56:51 Capturing 1.000-second  image...
2019-03-16T13:56:51 Job requires 1.000-second  images, has 0/1 frames captured and will be processed.

Same when I run the qhy_ccd_test program as root (indiserver & ekos stopped):

QHYCCD|QHY9S.CPP|GetSingleFrame|  GetSingleFrame   END   3584 2574 16 1
QHYCCD|QHY9S.CPP|GetSingleFrame|  GetSingleFrame   START
QHYCCD|QHYCAM.CPP|readUSB2B|readUSB2B p_size=16384 p_num=1127
QHYCCD|QHYCAM.CPP|readUSB2B|i=0
QHYCCD|QHYCAM.CPP|readUSB2B|ERROR libusb_bulk_transfer readUSB2B

Maybe I have a problem in the usb setup? The ccd is attached directly to the computer.

/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=ehci-pci/5p, 480M
    |__ Port 5: Dev 2, If 0, Class=Vendor Specific Class, Driver=, 480M

There is no driver attached. Is this normal?

The  libqhy package installed the file /lib/udev/rules.d/85-qhyccd.rules , which does not include a line for 1618:8301, only for 1618:8300:

ATTRS{idVendor}=="1618", ATTRS{idProduct}=="8300", RUN+="/sbin/fxload -t fx2 -I /lib/firmware/qhy/QHY9S.HEX -D $env{DEVNAME}"

I noticed, that a few days ago a new SDK has been released, which seems to include a line for 8301:
ACTION=="add", SUBSYSTEM=="usb", ATTRS{idVendor}=="1618", ATTRS{idProduct}=="8301", MODE="0666", GROUP="plugdev"

So perhaps this is missing and the reason why no driver is attached? 

Best, Peter

peter-englmaier, 2019-03-17:
Hi Jasem
I got it working! When I access the camera in "local mode", the camera works fine. The problem is only when I access it indirectly using the indiserver running in the same machine under systemd. It works, however, when I start the indiserver in my regular user (without systemd). There is probably a correct way to start it with systemd. 
It also printed error messages that it could not access locations like "(null)/.indi/..." which probably means the HOME environment variable is not set when started with systemd.  Maybe indiserver could print some error message when started in such a way.
Best, Peter

closed 2019-03-17 by peter-englmaier