indilib/indi #892: indi_toupcam_ccd crash on startup

derbuur, 2019-03-21:
indi_toupcam_ccd crash on startup.

INDI Server:  "2019-03-21T20:00:57: Driver indi_toupcam_ccd: indi_toupcam_ccd: symbol lookup error: indi_toupcam_ccd: undefined symbol: Toupcam_get_BlackBalance"

log attached
[log_21-00-48.txt](https://github.com/indilib/indi/files/2993767/log_21-00-48.txt)

rmathar, 2019-04-09:
Apparently indilib has been compiled with the request WITH_TOUPCAM but the actual driver (the touptek SDK library) was not found at compile time, so the function Toupcam_get_BlackBalance() is not linked with the executable. So the recommendation is to make sure that the library is actually found when indilib is compiled.

knro, 2019-04-23:
Any feedback on this?

rmathar, 2019-05-13:
If I were the original poster, I would at least do the following:
check that the touptek library is still present at the place where it was when the indilib was compiled. If it still is there check that environment variables like LD_LIBRARY_PATH include its directory. Then run the commands like nm and objdump on that library and see whether the symbol Toupcam_get_BlackBalance is in the table of resolved symbols. If it is not, upgrade the ToupTek library to a newer release such that the symbol exists. I could not find release notes in the ToupTek library which indicate when the unresolved function was introduced in the library.

knro, 2019-06-22:
Closing due to lack of feedback.

closed 2019-06-22 by knro