indilib/indi #897: QHYCFW2 can't set number of filters

RCompassi, 2019-03-28:
Can't set the number of filters on standalone QHYCFW2. Using the number of filters slider on the main control tab to set and save it to 7 does nothing, upon connecting the FW it returns to the default 5 slots.

knro, 2019-03-29:
What's the driver version for QHYCFW2?

RCompassi, 2019-03-29:
In the drive Info, it says 1.0.

RCompassi, 2019-03-29:
Ok, managed to update the qhy driver for 1.2.

Ok, here's what I've done:
Pulled the indilib from github
compiled the library
compiled the 3rd party driver indi-qhy mannualy and installed
THEN it updated to driver version 1.2, and now i can set the number of filter slots correctly.
For some reason, the nightly does not update the driver?

knro, 2019-03-30:
Nightly can get delayed due to launchpad issues sometimes. Issue is resolved, thanks!

closed 2019-03-30 by knro