indilib/indi #905: [indi_starbook_telescope] Can't park mount

not7cd, 2019-04-04:
Can't park mount using INDI with selecting *Park* switch. The same happens when user goes to http://starbook-ip/HOME, endpoint returns error. Telescope can be returned to HOME manually using Starbook physical interface.

**Steps to reproduce**
Exact steps to reproduce the behavior:
1. Run driver
2. Press *Park* in *Main control* tab inside INDI control panel
3. Starbook returns ERROR_FORMAT

**Expected behavior**
Starbook should return OK response and mount should slew to HOME position.

**System info:**
 - OS: raspbian
 - indi_starbook_telesckope 0.5
 - indiserver: 1.7.4
 - Starbook firmware version: 2.7

lkcheng89, 2019-04-09:
The correct command should be "GOHOME?HOME=0"

not7cd, 2019-04-09:
Will test this in the next occasion with Starbook L (1). I haven't found 
any string matching GOHOME in hexdump of latest 2.7 update executable.

If that's the case, it's not matching with documentation for Starbook 
1.2. and it might be first breaking change since then.

lkcheng89, 2019-04-10:
This is adapted from farhi/matlab-starbook repository of github. Indeed it works when sending GOHOME?HOME=0 with my Starbook S. ERROR:FORMAT is received when sending HOME which is described in documentation.

commit 13fa07a0 (2019-04-17): Fix HOME command

Resolve #905
closed 2019-04-17 by knro