indilib/indi #920: AAGCloudWatcher is not a subclass of indiweather

FrazCake, 2019-04-19:
Looking at the source code i noticed that AAGCloudWatcher is subclassing "DefaultDevice" instead of Weather class. I think that if possible it should be updated because in this way it seems to me that it can't be used to do autopark operations.

knro, 2019-04-20:
Correct, it was written LONG ago and needs to be migrated to INDI::Weather.. any volunteers? ;-)

pardo-bsso, 2019-05-21:
We are using one of these and it behaves kind of erratically from time to time, so I'm in. 
But I woulnd't mind more hands.

knro, 2019-05-21:
I can help, but there are too many parameters so it would need careful testing. We can create a branch for the migration to INDI::Weather

pardo-bsso, 2019-05-23:
Ok, I'll start this week with the migration at https://github.com/telescopio-montemayor/indi/tree/aagcloudwatcher-update

knro, 2019-06-22:
How is that going?

pardo-bsso, 2019-06-23:
On Sat, 22 Jun 2019 at 15:22, Jasem Mutlaq <notifications@github.com> wrote:
>
> How is that going?
>

Hey,
we got sidetracked with more pressing needs at our observatory and I
also lost remote access to the station. I'm trying to bring it home
but things are slow here at the university.

pardo-bsso, 2019-07-03:
Hi @knro , @FrazCake and anybody else using a cloudwatcher.

Can you please try the driver at https://github.com/telescopio-montemayor/indi/tree/aagcloudwatcher-update ?

I'm still not happy with the overall shape of the driver but works here. I have a couple of parameters to add yet.

Also, if anyone wants to donate/lend a pocket cloud watcher or the rh sensor it would be great.

knro, 2019-07-03:
Thanks, great progress! Unfortunately, I don't have AAG so I can't test anything now :(

pardo-bsso, 2019-07-04:
I only have an older model.
But I could use a bit of code review.

And also, if anyone wants to ask for features this would be a good time to do so.

FrazCake, 2019-07-06:
Hi!
Thank you for all the jobs you are doing!

> Can you please try the driver at https://github.com/telescopio-montemayor/indi/tree/aagcloudwatcher-update ?

I have tried it with kstars today but it can't connect. I have set the correct port but the connect led stay yellow while the log says it was connected.

`2019-07-06T12:55:08: Driver indi_aagcloudwatcher_ng: read message AAG Cloud Watcher NG   '[INFO] Connected to AAG Cloud Watcher'
<message device="AAG Cloud Watcher NG" timestamp="2019-07-06T12:55:08" message="[INFO] Connected to AAG Cloud Watcher"/>
`

I hope that log file can help.
As for the cloudwatcher we have bought it in 2017 and the firmware version is 5.60 (it's a strange thing because it seems that the last one is 5.51, so i will downgrade as soon as i can and test it again). 
I have also tried the old driver to be sure that cable is in good condition and it works.

> Also, if anyone wants to donate/lend a pocket cloud watcher or the rh sensor it would be great.

Unfortunally it's not mine but it belongs to a non profit observatory where i'm a volunteer.

[aag_log.txt](https://github.com/indilib/indi/files/3364890/aag_log.txt)

pardo-bsso, 2019-07-08:
@FrazCake 

> I have tried it with kstars today but it can't connect. I have set the correct port but the connect led stay yellow while the log says it was connected.
>
> `2019-07-06T12:55:08: Driver [...]

That's odd. I see nothing strange on the logfile.
I had similar issues with a CH340 usb adapter with both the original driver and this one, it needed to be connected and disconnected a couple of times until it worked.

> As for the cloudwatcher we have bought it in 2017 and the firmware version is 5.60 (it's a strange thing because it seems that the last one is 5.51, so i will downgrade as soon as i can and test it again).

The only one we have is the same version and around same time of manufacture.

> I have also tried the old driver to be sure that cable is in good condition and it works.

I'm reusing the protocol part almost as is, but now that I see I left over some parts specific to the serial port configuration.

> > Also, if anyone wants to donate/lend a pocket cloud watcher or the rh sensor it would be great.
> 
> Unfortunally it's not mine but it belongs to a non profit observatory where i'm a volunteer.

Same here, until I have this working fully again I can monitor the weather in my kitchen.

Could you please try the test program aagcloudwatcher_test_ng ? It's hardcoded to /dev/ttyUSB0 in the file main.cpp if you need to change it.

FrazCake, 2019-07-11:
I tried again and finally found the reason of the strange behaviour.
It seems that if i use a remote indiserver it doesn't connect, otherwise if i use it directly on the PC where Kstars is running it works. I have also test with aagcloudwatcher_test_ng and it was able to detect the device. If you can, just try to use kstars on a pc and indiserver on another one to replicate the bug.
In the coming weeks I will test the functionality to see if it can read the sensors correctly, close when it rains, etc.

pardo-bsso, 2019-07-11:
> I tried again and finally found the reason of the strange behaviour.
> It seems that if i use a remote indiserver it doesn't connect, otherwise if i use it directly on the PC where Kstars is running [...]

That's quite odd. I'll give it a go tonight.
Do you have different library versions between machines? Here I run on
the terminal

indiserver -vv ./indi_aagcloudwatcher_ng

And then connect to it as a client from Kstars over tcp, so it
shouldn't (in theory) be different from your scenario. But I've been
spectacularly wrong in the past.

pardo-bsso, 2019-07-14:
Hi again.

I'm not able to reproduce that using indi from master (6b5ebd0dbf36c70549d09c2d7a10c07193fbaa4b) and kstars 3.3.1

Is there anything else I might be missing to reproduce it?

FrazCake, 2019-07-14:
Hi!
Yesterday other strange things happened. We were making some observations and when I plugged in the cloudwatcher to do some tests the telescope stopped working because it couldn't transmit on /dev/ttyUSB0. The strange thing is that the telescope was connected to a Rasperry pi while the cloudwatcher was on the pc. Maybe it was a coincidence though.
I am attaching a screenshot to give you an idea of ​​how local and remote servers are connected (in Lan in this case).
I also have some bugs with the 3.3.2 version of kstars because all the objects of solar system have wrong coordinates, so maybe it could be just my system that has some problems. Now I have to stop testing and keep things stable because we will have many openings, I will try better in August, when our observatory will be closed for one month. I'll try to reinstall Ubuntu and recompile everything with same libraries and so on, but probably i have to wait till August to proceed. I'll let you know as soon as i can.

pardo-bsso, 2019-07-15:
Let's get weird then. 

This doesn't ring a bell to me but perhaps @knro can give us some insight or at least bring more users to this discussion.

knro, 2019-07-17:
@FrazCake The bug with wrong positions in non-English KStars is fixed in GIT.

The strange behavior you encountered is odd, but not related to how your start the profile. Perhaps at this stage, it might be better to start a profile with only the telescope+ccd simulator in addition to the under-development cloudwatcher driver? Once it is more mature, it can be used with actual devices to further test its behavior.

FrazCake, 2019-07-17:
> The bug with wrong positions in non-English KStars is fixed in GIT.

Oh ok, i changed to English and now it works, thank you!

> Perhaps at this stage, it might be better to start a profile with only the telescope+ccd simulator in addition to the under-development cloudwatcher driver?

I will try this way as soon as i can :thumbsup:

knro, 2019-09-16:
Hello, is there any update on this?

knro, 2019-09-18:
Thanks for the report. Please open this issue in the 3rdparty repo issues --> https://github.com/indilib/indi-3rdparty/issues

closed 2019-09-18 by knro