indilib/indi #921: Indilib 1.7.6 has changed moving direction of Sesto Senso focuser.

jogijo, 2019-04-20:
My Sesto Senso focuser is moving in the wrong direction since the release of Indilib 1.7.6, which I use with StellarMate OS on a Raspberry Pi.

Everything was working perfectly with Indilib 1.7.5 (on StellarMate OS 1.3.1). But since Indilib 1.7.6 (with StellarMate OS 1.3.2) the focuser suddenly behaved weird and moved in the opposite direction than it did before.

The new feature "Reverse Motion" introduced with Indilib 1.7.7 does not really solve the problem. In fact it does change the moving direction but the absolute focuser position in the driver is counting wrong. It counts down when moved out and counts up when moving in. It must be the other way around. This leads to a new weird behaviour. When you start with absolute position 0 (focuser completely retracted) and you try to move out it writes to the log "Focuser reached requested position." and it does not move. Moving in it writes to the log "Requested position out of bound. Focus minimum position is 0" it does not move either, which is correct in this case. Really moving the motor is only possible with the focuser positioned somewhere and syncing the driver to some arbitrary position. But it does not help since the position is counting wrong.
 
It also seems that the calibration which was done with the original Primaluce Lab software was broken, because after using the focuser only once with Indilib 1.7.6 (StellarMate OS 1.3.2) it wouldn't work any longer with Indilib 1.7.5 (StellarMate OS 1.3.1) either.

Something fundamental must have been changed from Indilib 1.7.5. to Indilib 1.7.6 which broke the behaviour of the focuser completely and made it unusable.

BTW: My focuser is mounted on the right side of a Sky-Watcher Evostar 72 ED. That's where the 1:11 focus knob is. This means that during initialization with the original Primaluce Lab software I have to teach the Sesto Senso focuser that the default moving direction has to be inverted to match the movement of my telescope's focuser. 

But anyway, it was working and now it's sadly broken.

 - OS: StellarMate OS 
 - Version: 1.3.2+ (Indilib 1.7.6+)
 - Sesto Senso Firmware: 2.1

knro, 2019-04-23:
Can you use the nightly release and test? Also remove the existing config from ~/.indi or use the new "Purge" configuration switch in INDI.

jogijo, 2019-04-23:
I've tested the nightly release as you advised. But nothing changed. It still counts in the wrong direction and does not move if retracted. It's unfortunately still as described in my initial post.

But there's one thing. Please forget what I have written after BTW. That was completely wrong. It moves out by pressing + and moves in by pressing -

rishi-garrod, 2019-06-02:
Can you try the nightly release again. You should get 1.3 of the driver which should fix the problem.

jogijo, 2019-06-03:
Thanks for the reply. I've tried to install the nightly build as described here: https://indilib.org/download.html

Unfortunately I still end up with driver version 1.2. I made two attempts. One with StellarmateOS 1.3.3 and one with 1.4.0. Both did not work.

Did I do something wrong?

rishi-garrod, 2019-06-03:
Hi 

Have you tried recently? 

I saw a message on the forum from someone who had sucessfully pulled the
nightly build and got version 1.3. 

Regards 

Rishi

 

Links:
------
[1]
https://github.com/indilib/indi/issues/921?email_source=notifications&amp;email_token=AMHF442ZAO2GSOR3IRT6I53PYV3ELA5CNFSM4HHKJ2M2YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGODW2RRQI#issuecomment-498407617
[2]
https://github.com/

jogijo, 2019-06-03:
Yes. I‘ve tried eight before I wrote my message. Hmmm... I will try again tomorrow.

jogijo, 2019-06-03:
OK. I had to try it this evening. I got it done. Getting the nightly built with the commands from the Stellarmate Website worked. And with the driver version 1.3 the Sesto Senso also works as expected. Great job! Thank you... :-)

closed 2019-06-05 by knro