indilib/indi #925: Open dome shutter can send a park command to the mount

FrazCake, 2019-04-21:
At line [552](https://github.com/indilib/indi/blob/3ba1e96f7efeffe971df636e01673108be5445ca/libindi/libs/indibase/inditelescope.cpp#L552) in inditelescope.cpp it is checked in the if statement that the propName is "DOME_SHUTTER" but this can trigger a park event to the mount if you open the shutter when mount is already unparked. This is because it is not checked that the elemName is "SHUTTER_CLOSE", so even a "SHUTTER_OPEN" will trigger a park event.

Don't know if it is a good solution but i have solved this way in my observatory:

Wrap from line [559](https://github.com/indilib/indi/blob/3ba1e96f7efeffe971df636e01673108be5445ca/libindi/libs/indibase/inditelescope.cpp#L559) to 561 like this
```
for (ep = nextXMLEle(root, 1); ep != nullptr; ep = nextXMLEle(root, 0))
{
    const char * elemName = findXMLAttValu(ep, "name");
    if ((!strcmp(elemName, "SHUTTER_CLOSE") && !strcmp(pcdataXMLEle(ep), "On")))
    {
------------------------------------[line 559]--------------------------------------
        RememberTrackState = TrackState;
        Park();
        DEBUG(Logger::DBG_SESSION, "Dome is closing, parking mount...");
------------------------------------[line 561]--------------------------------------
    }
}
```

knro, 2019-04-22:
shouldn't this includes a check for PARK element in DOME_PARK as well?

if (( (!strcmp(elemName, "SHUTTER_CLOSE") || !strcmp(elemName, "PARK")) && !strcmp(pcdataXMLEle(ep), "On")))

FrazCake, 2019-04-22:
Oh, yes it must be included. I didn't notice because i always close shutter on park, but now it seems to me that it is necessary. Otherwise if for any reason you park the dome without closing the shutter, the mount would not receive the park command.

commit 5367de8a (2019-04-22): Check if either the shutter closing or dome parking to trigger mount parking if enabled.

Fixes #925
closed 2019-04-22 by knro