indilib/indi #928: Indi_qhy does not compile

azwing, 2019-04-26:
**Describe the bug**
When building 3rdparty drivers indi_qhy fails with following message:
=====================================================
rdparty/indi-qhy/qhy_ccd.cpp:611:56: error: ‘CAM_LIVEVIDEOMODE’ was not declared in this scope
         ret = IsQHYCCDControlAvailable(m_CameraHandle, CAM_LIVEVIDEOMODE);
                                                        ^~~~~~~~~~~~~~~~~
/3rdparty/indi-qhy/qhy_ccd.cpp:611:56: note: suggested alternative: ‘CAM_VIEW_MODE’
         ret = IsQHYCCDControlAvailable(m_CameraHandle, CAM_LIVEVIDEOMODE);
=====================================================
 OS Release
PRETTY_NAME="Debian GNU/Linux buster/sid"
NAME="Debian GNU/Linux"

knro, 2019-04-26:
make sure latest libqhy is installed first.

azwing, 2019-04-26:
I removed all previous libraries before compiling 
by issuing "sudo xargs rm < install_manifest.txt " 
then compile / install the library 
but I still have the same issue

knro, 2019-04-27:
was latest libqhy installed from GIT?

azwing, 2019-04-27:
yes! I work like this since long time.
1) Git pull
2) Build / Install libindi
3) Build / Install 3rdparty
When I get compilation errors then I remove all libs and rebuild all from scratch.
I willm redo all the thing again and let you know.

azwing, 2019-04-27:
Finally I got it .....

there was a Include file under /usr/local/include where it should not be at all !!!!
Shame on me !

closed 2019-04-27 by azwing