indilib/indi #943: SkySafari 6 Pro will not connect with Synscan mount

KjetilLe, 2019-05-06:
I can connect with SkySafari 6 Pro when I select Telescop Simulator as Mount in Indi.
However, when I scange the mount to Synscan (connected through USB to serial cable), SkySafari is unable to connect. I get the message "SkySafari Pro can't make a wireless connection to the scope. If you're using a bluetoth serial adapter, make sure that it's turned on and paired with the Android device. If you're using wifi, make sure you've joined the scope's Wi-Fi, and entered its IP address correctly."

In Indi control panel I get the message "SkySafari Server is running. Connect the App now to this machine using SkySafari LX200 driver. 
(In SkySafari I have selected Meade LX200 GPS/ACF, LX600 driver.) 
(I have also tried the others Meade mounts in SkySafari with no luck.) 

K-stars is working fine, the problem is only to Connect SkySafari.

KjetilLe, 2019-05-06:
I use Stellarmate version 1.3.5 and the software updater says I am up to date.

KjetilLe, 2019-05-07:
Thank you again for Your fast reply, but how do I incorporate this in the system?

Best regards
Kjetil

Closed #943<https://eur03.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2Findilib%2Findi%2Fissues%2F943&data=02%7C01%7C%7C4b3e4262f5184baf0e9708d6d2d091f2%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C636928190567689507&sdata=3OQYZfuXkCYHCSKLUlyT3Hg8CcrN9yM2tOJ7aOPA%2BZA%3D&reserved=0> via 5bac720<https://eur03.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2Findilib%2Findi%2Fcommit%2F5bac72018ca3ef5e721c4e301faf36ba8c375579&data=02%7C01%7C%7C4b3e4262f5184baf0e9708d6d2d091f2%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C636928190567699512&sdata=F7Ha%2B0L8BYs6l3Kua5EVMmh1e0Uty53UfHUaLA3xH1k%3D&reserved=0>.

TallFurryMan, 2019-05-07:
Shift your stellarmate to pick updates from nightly builds with "sudo add-apt-repository ppa:mutlaqja/indinightly", update once the mentioned commit is built by Jasem's ppa.

knro, 2019-05-07:
It's not yet available, I'm still working on the client part (KStars) since it needs to be updated. This is not really a "fix" it is just convenience. The issue is that the Snoop Telescope property in Options in the SkySafari driver is not set to your mount, so now I'm updating the client (KStars) to do this automatically. But you can do it yourself once and just change the field from Telescope Simulator to Synscan or whatever mount you're using and it should be set.

KjetilLe, 2019-05-08:
Hi

I tried again this morning, but it is still not working. I enclose the log.
 In SkySafari I get message: "SkySafari Pro can make a wireless connection to the scope, but the scope is not responding. Make sure the scope is powered on and connected correctly. Also make sure that you select the correct scope type."

In K-star: "[Error] Internal client is not connected! Please make sure the mount name is set in the Options tab. Disconnect and reconnect to try again."

I hope this helps to solve the problem.

Best regards
Kjetil

[2019-05-08T11:56:52.732 +03 DEBG ][           org.kde.kstars.ekos] - Resetting Ekos Manager...
[2019-05-08T11:56:52.749 +03 WARN ][                       default] - QSqlDatabasePrivate::removeDatabase: connection 'filter_db' is still in use, all queries will cease to work.
[2019-05-08T11:56:52.749 +03 WARN ][                       default] - QSqlDatabasePrivate::addDatabase: duplicate connection name 'filter_db', old connection removed.
[2019-05-08T11:56:52.860 +03 DEBG ][           org.kde.kstars.indi] - INDI: Starting local drivers...
[2019-05-08T11:56:52.861 +03 DEBG ][           org.kde.kstars.indi] - Starting INDI Server:  ("-v", "-p", "7624", "-f", "/tmp/indififoc3cd1813") -f "/tmp/indififoc3cd1813"
[2019-05-08T11:56:52.900 +03 DEBG ][           org.kde.kstars.indi] - INDI Server Started?  true
[2019-05-08T11:56:52.901 +03 DEBG ][           org.kde.kstars.indi] - INDI: INDI Server started locally on port  7624
[2019-05-08T11:56:52.910 +03 DEBG ][           org.kde.kstars.indi] - Starting INDI Driver  "indi_synscan_telescope"
[2019-05-08T11:56:52.910 +03 DEBG ][           org.kde.kstars.indi] - Starting INDI Driver  "indi_simulator_ccd"
[2019-05-08T11:56:52.911 +03 DEBG ][           org.kde.kstars.indi] - Starting INDI Driver  "indi_gpsd"
[2019-05-08T11:56:52.912 +03 DEBG ][           org.kde.kstars.indi] - Starting INDI Driver  "indi_skysafari"
[2019-05-08T11:56:52.913 +03 DEBG ][           org.kde.kstars.indi] - Adding managed driver "SynScan"
[2019-05-08T11:56:52.913 +03 DEBG ][           org.kde.kstars.indi] - Adding managed driver "CCD Simulator"
[2019-05-08T11:56:52.914 +03 DEBG ][           org.kde.kstars.indi] - Adding managed driver "GPSD"
[2019-05-08T11:56:52.914 +03 DEBG ][           org.kde.kstars.indi] - Adding managed driver "SkySafari"
[2019-05-08T11:56:52.915 +03 DEBG ][           org.kde.kstars.indi] - INDIListener: Adding a new client manager to INDI listener..
[2019-05-08T11:56:52.916 +03 DEBG ][           org.kde.kstars.indi] - INDI: Connecting to local INDI server on port  7624  ...
[2019-05-08T11:56:52.918 +03 DEBG ][           org.kde.kstars.indi] - INDI server connected.
[2019-05-08T11:56:52.918 +03 DEBG ][           org.kde.kstars.indi] - Connection to INDI server is successful
[2019-05-08T11:56:52.943 +03 DEBG ][           org.kde.kstars.indi] - Received new device SkySafari
[2019-05-08T11:56:52.947 +03 DEBG ][           org.kde.kstars.indi] - INDI Server:  "2019-05-08T08:56:52: startup: /usr/bin/indiserver -v -p 7624 -f /tmp/indififoc3cd1813 "
[2019-05-08T11:56:52.948 +03 DEBG ][           org.kde.kstars.indi] - INDI Server:  "2019-05-08T08:56:52: listening to port 7624 on fd 3"
[2019-05-08T11:56:52.948 +03 DEBG ][           org.kde.kstars.indi] - INDI Server:  "FIFO: start indi_synscan_telescope -n \"SynScan\""
[2019-05-08T11:56:52.948 +03 DEBG ][           org.kde.kstars.indi] - INDI Server:  "With name: SynScan"
[2019-05-08T11:56:52.948 +03 DEBG ][           org.kde.kstars.indi] - INDI Server:  "FIFO: Starting driver indi_synscan_telescope"
[2019-05-08T11:56:52.948 +03 DEBG ][           org.kde.kstars.indi] - INDI Server:  "2019-05-08T08:56:52: Driver indi_synscan_telescope: pid=13124 rfd=4 wfd=7 efd=8"
[2019-05-08T11:56:52.949 +03 DEBG ][           org.kde.kstars.indi] - INDI Server:  "FIFO: start indi_simulator_ccd"
[2019-05-08T11:56:52.949 +03 DEBG ][           org.kde.kstars.indi] - INDI Server:  "FIFO: Starting driver indi_simulator_ccd"
[2019-05-08T11:56:52.949 +03 DEBG ][           org.kde.kstars.indi] - INDI Server:  "2019-05-08T08:56:52: Driver indi_simulator_ccd: pid=13125 rfd=5 wfd=10 efd=11"
[2019-05-08T11:56:52.949 +03 DEBG ][           org.kde.kstars.indi] - INDI Server:  "FIFO: start indi_gpsd -n \"GPSD\""
[2019-05-08T11:56:52.949 +03 DEBG ][           org.kde.kstars.indi] - INDI Server:  "With name: GPSD"
[2019-05-08T11:56:52.950 +03 DEBG ][           org.kde.kstars.indi] - INDI Server:  "FIFO: Starting driver indi_gpsd"
[2019-05-08T11:56:52.950 +03 DEBG ][           org.kde.kstars.indi] - INDI Server:  "2019-05-08T08:56:52: Driver indi_gpsd: pid=13126 rfd=6 wfd=13 efd=14"
[2019-05-08T11:56:52.950 +03 DEBG ][           org.kde.kstars.indi] - INDI Server:  "FIFO: start indi_skysafari -n \"SkySafari\""
[2019-05-08T11:56:52.950 +03 DEBG ][           org.kde.kstars.indi] - INDI Server:  "With name: SkySafari"
[2019-05-08T11:56:52.950 +03 DEBG ][           org.kde.kstars.indi] - INDI Server:  "FIFO: Starting driver indi_skysafari"
[2019-05-08T11:56:52.950 +03 DEBG ][           org.kde.kstars.indi] - INDI Server:  "2019-05-08T08:56:52: Driver indi_skysafari: pid=13127 rfd=9 wfd=19 efd=20"
[2019-05-08T11:56:52.951 +03 DEBG ][           org.kde.kstars.indi] - INDI Server:  "2019-05-08T08:56:52: Client 12: new arrival from 127.0.0.1:32868 - welcome!"
[... cut ...]
[2019-05-08T11:57:25.926 +03 DEBG ][           org.kde.kstars.indi] - SynScan : "[DEBUG] RES <0#> "
[2019-05-08T11:57:25.926 +03 DEBG ][           org.kde.kstars.indi] - SynScan : "[DEBUG] CMD <p> "
[2019-05-08T11:57:25.962 +03 DEBG ][           org.kde.kstars.indi] - SynScan : "[DEBUG] RES <W#> "
[2019-05-08T11:57:25.963 +03 DEBG ][           org.kde.kstars.indi] - SynScan : "[DEBUG] CMD <t> "
[2019-05-08T11:57:25.992 +03 DEBG ][           org.kde.kstars.indi] - SynScan : "[DEBUG] RES <#> "
[2019-05-08T11:57:25.993 +03 DEBG ][

knro, 2019-05-08:
Did you check SkySafari INDI driver ---> Options --> Snoop Telescope

That should be set the the mount name you're using. Then click "Save" after that. Then disconnect/reconnect.

KjetilLe, 2019-05-08:
Hi

My first attempt was With synscan, but changing to Snoop telescope does not help.

Shall I make a New log file?

Best regards
Kjetil

knro, 2019-05-08:
Please attach a screenshot of the SkySafari driver Options tab.

KjetilLe, 2019-05-08:
Here it is.

Best regards
Kjetil

KjetilLe, 2019-05-19:
[image]
[image]

KjetilLe, 2019-05-19:
Above is the screenshot you requested some days ago.
Regarding the Bluetooth Connection: I connect the Bluetooth dongle from the Bluetooth menue on the main screen shown on the photo and I get a connection. Then I Guess the baud rate is agreed upon (I am no expert here). However I do not get a flashing yellow ligt in the dongle (as I get when I onnect to the mount via Bluetooth directly (without Stellarmate) from SkySafari 6 Pro to the mount (chosing SkyWatcher SynScan in the SkySafari menue). When I do this I have a steady green light and a flashing yellow light at the Bluetooth dongle.
The dongle is connected when I open and start Kstars, but the Bluetooth Connection will not Connect.
Before investigating further I would like to know that this is the correct procedure. In Ekos I can not select the Serial port Assistant and that confuses me.

KjetilLe, 2019-05-19:
I have tried all the Baud rates, but I get the Warning that the port/dev/rfcomm0 is already used by another driver or process no matter which Baud rate I use.

KjetilLe, 2019-07-04:
SkySafari connects now with Stellarmate 1.4.2 using EQMod mode.

commit 5bac7201 (2019-05-07): Define standard properties for SkySafari driver. Fixes #943 in KStars v3.2.2
closed 2019-07-04 by KjetilLe