indilib/indi #944: Bluetooth will not connect

KjetilLe, 2019-05-06:
I can connect to my Synscan using an USB to serial cable, but I can not connect using a Bluetooth dongle (Firefly). The Bluetooth device connects and I get up- and downloads.

When scanning ports I get 3: /dev/rfcomm0, /dev/ttyACM0 and /dev/ttyUSB0.
None of the ports will connect.
When using /dev/rfcomm0, which I believe is the correct one, I get the following error message: [Warning] Port /dev/rfcomm0 is already used by another driver or process. I get the same message using /dev/ttyACM0. For dev/ttyUSB0 I get the message [Error] Serial read error: Timeout error.

I use Stellarmate version 1.3.5 and the software updater says I am up to date.

knro, 2019-05-07:
/dev/rfcomm0 is indeed the correct one. Can you disconnect all other devices and just keep the bluetooth to see if that works?

KjetilLe, 2019-05-07:
Thank you for the quick reply.
I have tried as you proposed and here is the logfile. (I had to add a CCD to be able to run.)

[2019-05-07T11:43:10.642 +03 DEBG ][           org.kde.kstars.ekos] - Resetting Ekos Manager...
[2019-05-07T11:43:54.379 +03 DEBG ][           org.kde.kstars.ekos] - Resetting Ekos Manager...
[2019-05-07T11:43:54.385 +03 WARN ][                       default] - QSqlDatabasePrivate::removeDatabase: connection 'filter_db' is still in use, all queries will cease to work.
[2019-05-07T11:43:54.385 +03 WARN ][                       default] - QSqlDatabasePrivate::addDatabase: duplicate connection name 'filter_db', old connection removed.
[2019-05-07T11:43:54.523 +03 DEBG ][           org.kde.kstars.indi] - INDI: Starting local drivers...
[2019-05-07T11:43:54.524 +03 DEBG ][           org.kde.kstars.indi] - Starting INDI Server:  ("-v", "-p", "7624", "-f", "/tmp/indififo0a876cf5") -f "/tmp/indififo0a876cf5"
[2019-05-07T11:43:54.561 +03 DEBG ][           org.kde.kstars.indi] - INDI Server Started?  true
[2019-05-07T11:43:54.561 +03 DEBG ][           org.kde.kstars.indi] - INDI: INDI Server started locally on port  7624
[2019-05-07T11:43:54.562 +03 DEBG ][           org.kde.kstars.indi] - Starting INDI Driver  "indi_synscan_telescope"
[2019-05-07T11:43:54.563 +03 DEBG ][           org.kde.kstars.indi] - Starting INDI Driver  "indi_simulator_ccd"
[2019-05-07T11:43:54.564 +03 DEBG ][           org.kde.kstars.indi] - Adding managed driver "SynScan"
[2019-05-07T11:43:54.565 +03 DEBG ][           org.kde.kstars.indi] - Adding managed driver "CCD Simulator"
[2019-05-07T11:43:54.567 +03 DEBG ][           org.kde.kstars.indi] - INDIListener: Adding a new client manager to INDI listener..
[2019-05-07T11:43:54.568 +03 DEBG ][           org.kde.kstars.indi] - INDI: Connecting to local INDI server on port  7624  ...
[2019-05-07T11:43:54.570 +03 DEBG ][           org.kde.kstars.indi] - INDI server connected.
[2019-05-07T11:43:54.570 +03 DEBG ][           org.kde.kstars.indi] - Connection to INDI server is successful
[2019-05-07T11:43:54.598 +03 DEBG ][           org.kde.kstars.indi] - Received new device CCD Simulator
[2019-05-07T11:43:54.614 +03 DEBG ][           org.kde.kstars.indi] - INDI Server:  "2019-05-07T08:43:54: startup: /usr/bin/indiserver -v -p 7624 -f /tmp/indififo0a876cf5 "
[2019-05-07T11:43:54.615 +03 DEBG ][           org.kde.kstars.indi] - INDI Server:  "2019-05-07T08:43:54: listening to port 7624 on fd 3"
[2019-05-07T11:43:54.615 +03 DEBG ][           org.kde.kstars.indi] - INDI Server:  "FIFO: start indi_synscan_telescope -n \"SynScan\""
[2019-05-07T11:43:54.615 +03 DEBG ][           org.kde.kstars.indi] - INDI Server:  "With name: SynScan"
[2019-05-07T11:43:54.615 +03 DEBG ][           org.kde.kstars.indi] - INDI Server:  "FIFO: Starting driver indi_synscan_telescope"
[2019-05-07T11:43:54.616 +03 DEBG ][           org.kde.kstars.indi] - INDI Server:  "2019-05-07T08:43:54: Driver indi_synscan_telescope: pid=4948 rfd=4 wfd=7 efd=8"
[2019-05-07T11:43:54.616 +03 DEBG ][           org.kde.kstars.indi] - INDI Server:  "FIFO: start indi_simulator_ccd"
[2019-05-07T11:43:54.616 +03 DEBG ][           org.kde.kstars.indi] - INDI Server:  "FIFO: Starting driver indi_simulator_ccd"
[2019-05-07T11:43:54.616 +03 DEBG ][           org.kde.kstars.indi] - INDI Server:  "2019-05-07T08:43:54: Driver indi_simulator_ccd: pid=4949 rfd=5 wfd=10 efd=11"
[2019-05-07T11:43:54.616 +03 DEBG ][           org.kde.kstars.indi] - INDI Server:  "2019-05-07T08:43:54: Client 6: new arrival from 127.0.0.1:41504 - welcome!"
[2019-05-07T11:43:54.616 +03 DEBG ][           org.kde.kstars.indi] - INDI Server:  "2019-05-07T08:43:54: Driver indi_simulator_ccd: snooping on Telescope Simulator.EQUATORIAL_EOD_COORD"
[2019-05-07T11:43:54.617 +03 DEBG ][           org.kde.kstars.indi] - INDI Server:  "2019-05-07T08:43:54: Driver indi_simulator_ccd: snooping on Telescope Simulator.TELESCOPE_INFO"
[2019-05-07T11:43:54.617 +03 DEBG ][           org.kde.kstars.indi] - INDI Server:  "2019-05-07T08:43:54: Driver indi_simulator_ccd: snooping on Telescope Simulator.GEOGRAPHIC_COORD"
[2019-05-07T11:43:54.617 +03 DEBG ][           org.kde.kstars.indi] - INDI Server:  "2019-05-07T08:43:54: Driver indi_simulator_ccd: snooping on Focuser Simulator.ABS_ROTATOR_ANGLE"
[2019-05-07T11:43:54.617 +03 DEBG ][           org.kde.kstars.indi] - INDI Server:  "2019-05-07T08:43:54: Driver indi_simulator_ccd: snooping on CCD Simulator.FILTER_SLOT"
[2019-05-07T11:43:54.617 +03 DEBG ][           org.kde.kstars.indi] - INDI Server:  "2019-05-07T08:43:54: Driver indi_simulator_ccd: snooping on CCD Simulator.FILTER_NAME"
[2019-05-07T11:43:54.618 +03 DEBG ][           org.kde.kstars.indi] - INDI Server:  "2019-05-07T08:43:54: Driver indi_simulator_ccd: snooping on SQM.SKY_QUALITY"
[2019-05-07T11:43:54.618 +03 DEBG ][           org.kde.kstars.indi] - INDI Server:  "2019-05-07T08:43:54: Driver indi_simulator_ccd: snooping on Focuser Simulator.FWHM"
[2019-05-07T11:43:54.618 +03 DEBG ][           org.kde.kstars.indi] - INDI Server:  "2019-05-07T08:43:54: Driver indi_synscan_telescope: snooping on GPS Simulator.GEOGRAPHIC_COORD"
[2019-05-07T11:43:54.618 +03 DEBG ][           org.kde.kstars.indi] - INDI Server:  "2019-05-07T08:43:54: Driver indi_synscan_telescope: snooping on GPS Simulator.TIME_UTC"
[... cut ...]
[2019-05-07T11:45:10.355 +03 WARN ][                       default] - QSqlDatabasePrivate::removeDatabase: connection 'filter_db' is still in use, all queries will cease to work.
[2019-05-07T11:45:10.355 +03 WARN ][                       default] - QSqlDatabasePrivate::addDatabase: duplicate connection name 'filter_db', old connection removed.
[2019-05-07T11:54:07.697 +03 DEBG ][                org.kde.kstars] - 0 StarBlocks freed from StarBlockFactory
[2019-05-07T11:54:07.736 +03 WARN ][                       default] - QSqlDatabasePrivate::removeDatabase: connection 'userdb' is still in use, all queries will cease to work.

knro, 2019-05-18:
Did you try different baud rates?

KjetilLe, 2019-05-27:
Above is the screenshot you requested some days ago.
Regarding the Bluetooth Connection: I connect the Bluetooth dongle from the Bluetooth menue on the main screen shown on the photo and I get a connection. Then I Guess the baud rate is agreed upon (I am no expert here). However I do not get a flashing yellow ligt in the dongle (as I get when I onnect to the mount via Bluetooth directly (without Stellarmate) from SkySafari 6 Pro to the mount (chosing SkyWatcher SynScan in the SkySafari menue). When I do this I have a steady green light and a flashing yellow light at the Bluetooth dongle.
The dongle is connected when I open and start Kstars, but the Bluetooth Connection will not Connect.
Before investigating further I would like to know that this is the correct procedure. In Ekos I can not select the Serial port Assistant and that confuses me.

KjetilLe, 2019-05-27:
I have tried all the Baud rates, but I get the Warning that the port/dev/rfcomm0 is already used by another driver or process no matter which Baud rate I use.

knro, 2019-05-28:
Can you please try to see if this is an issue as in the latest SM OS v1.4.0?

KjetilLe, 2019-05-30:
Same result With v1.4.0. I can Connect via Serial cable but not via Bluetooth. 
I seems that the Bluetooth dongle Connects (to Serial port 6), but I get the Message that 6 is already in use.
[image]
[image]
[image]

KjetilLe, 2019-05-30:
[image]

knro, 2019-05-31:
Ok can you build libindi from source? If yes, I can suggest one patch that we can check if it works out for you. Or if you have TeamViewer setup on SM OS and ready, open a support ticket at StellarMate and we can arrange time for the TV setup.

KjetilLe, 2019-05-31:
Hi Jasem

I am not a libindi Builder, but TeamViewer can be arranged. However I will be away from home the Next 2-3 weeks doing a Hydro Power Station commissioning. I will contact you when I am back home.

best regards
Kjetil

knro, 2019-09-15:
v1.8.1 should have the fix. Can you see if it works with latest stable INDI?

KjetilLe, 2019-09-16:
Hi Jasem
I can confirm that is working with this update.
Than you.

Best regards
Kjetil

closed 2019-09-16 by knro