indilib/indi #950: ZWO ASI120MM Mini: USB reset on each frame

pludov, 2019-05-17:
Running an exposure with the ASI120MM Mini and the indi_asi_ccd consistently makes the linux kernel complain about usb reset ! Resetting an usb device indicates a stability problem and may affect others devices.

This happens on various hardware (linaro & laptop) with multiple USB cables. I also tried 16 vs 8bits and various bin and exposure duration with the same result.

However, it does not happen during video recording.

I also experienced other crashes of the indi driver and misbehavior (partial images dl), that may be related

Indi driver is v1.7.8.

The log following log is kernel output for plugin the camera then taking 3 shoots:

[16924.999689] usb 3-4.1: new high-speed USB device number 10 using xhci_hcd
[16925.100303] usb 3-4.1: New USB device found, idVendor=03c3, idProduct=120c, bcdDevice= 0.00
[16925.100309] usb 3-4.1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[16925.100312] usb 3-4.1: Product: ASI120MM Mini
[16925.100315] usb 3-4.1: Manufacturer: ZWO
[16948.843713] usb 3-4.1: reset high-speed USB device number 10 using xhci_hcd
[16954.259806] usb 3-4.1: reset high-speed USB device number 10 using xhci_hcd
[17209.970631] usb 3-4.1: reset high-speed USB device number 10 using xhci_hcd

knro, 2019-05-18:
This is most likely an issue with the SDK itself.. how would the INDI driver itself cause a USB reset? Maybe it's worthwhile compiling the driver against older SDKs to see if the issue manifests there?

pludov, 2019-05-19:
Thanks for your reply. What puzzle me is that it does not happen during streaming...

I tried with SDK v1.14.0227 and v1.14.0425 with the same result.
The debug log is clean, no mention of any exposure failed.

The usb reset always happens on exposure start. (I debbuged this with exposure > 1s...)

Maybe relevant: the cam is USB3 plugged on USB2 host. No "compatible" firmware available from ZWO.

pludov, 2019-05-19:
I also noted two other problems, while taking sequences (with covered scope, only dark signal expected):
 * Frames does not come regularely : in continuous mode with 1s exposure, they come by burst of two. When triggered manually,  they sometime arrive sooner than the requested exposure duration ! (just watch the evolution of the exposure duration for exp of >= 2s : for some photo, it does not go bellow 1s.
 * Some images are under exposed (would say about 1 per minute at 1s exposure). After dark substraction, those are almost totally black, meaning either a gain/level problem, or a shorter  exposure duration

pludov, 2019-06-02:
Hello 

any news on this subject ? I don't think that the ZWO linux sdk is public so I can't do more investigations... Do you have a contact with ZWO ?

knro, 2019-06-02:
I asked ZWO to look into it.

smeird, 2019-06-05:
Just to add , Im  also getting the same resets on linux ( ubuntu 18 & 19) also both of my cameras asi120mm & asi 183mm, frames work, but system slows becomes unstable. Ive even tested virtual hardware.

smeird, 2019-06-08:
I can provide logs if needed etc

knro, 2019-06-09:
This is happening on the SDK level, right? so if that's the case, then it would probably be better to talk to ZWO ASI directly over this.

pludov, 2019-06-10:
Discussion with ZWO  engaged here :
https://bbs.astronomy-imaging-camera.com/viewtopic.php?f=29&t=9937

smeird, 2019-06-13:
I received this from ZWO support , the reset is deliberate 
—
I checked the videos, but not very understand.
For the reset, now I am sure there is a reset, and I checked our program again, find it this time. This reset is used to resolve internal device exceptions. For example, the last time a unknown problem occurred, the device was not released, and the next time you use it again, the device will be busy. With this function reset, the device can be released, ensuring the next normal transmission. I am sure the reset you find in the log is this. For this reset, we just reset our device. It will not affect other devices.

knro, 2019-06-22:
Thanks for the update. So it's an upstream issue.

closed 2019-06-22 by knro