indilib/indi #953: Race condition when capturing guide frames, with possible fix

k-ross, 2019-05-25:
When capturing guide frames, I would occasionally get the error in the INDI panel that the requested value (0) was out of range. Also occasionally, I would get a really, really short exposure, which would result in a guide frame with just noise.

Digging in, and putting a lot of debug logging in place, I think the problem comes from when the INDI server sends back status messages about the current exposure remaining. The problem seems to be made worse because the server and client are communicating over WiFi, which slows down the responses from the server to the client.

So, I believe what is happening:

1. Guide module requests the CCD to capture a new frame.
2. CCDChip::capture gets the exposure INumberVectorProperty from the device, and updates the value to the new exposure value.
3. (I think this is where things go wrong) The client receives a notification from the server that the exposure value has changed, overwriting the INumberVectorProperty above
4. CCDChip::capture sends the wrong value to sendNewNumber()

I verified that # 3 is happening, by modifying CCDCHip::capture so that it prints the value of the INumberVectorProperty right after it is set, and loops a few times with a small delay, and prints the values again. During this time, the value will sometimes change all on its own to 0, leading me to believe that # 3 above is the culprit.

My fix is to modify CCDChip::capture to clone the INumberVectorProperty into a local variable and make changes there. Since I don't know the proper way to clone an INumberVectorProperty, my method is probably not ideal.

But here's my version of CCDChip::capture:

```
bool CCDChip::capture(double exposure)
{
    auto expProp = std::make_unique<INumberVectorProperty>();
    INumber n;
    expProp->np = &n;
    expProp->nnp = 1;
    strcpy(expProp->device, getCCD()->getDeviceName());

    switch (type)
    {
        case PRIMARY_CCD:
            strcpy(expProp->name, "CCD_EXPOSURE");
            strcpy(n.name, "CCD_EXPOSURE_VALUE");
            break;

        case GUIDE_CCD:
            strcpy(expProp->name, "GUIDER_EXPOSURE");
            strcpy(n.name, "GUIDER_EXPOSURE_VALUE");
            break;
    }

    n.value = exposure;

    clientManager->sendNewNumber(expProp.get());

    return true;
}

```

knro, 2019-05-25:
Thank you for the detailed investigation! This is a KStars issue, not INDI so it doesn't belong here (the correct place is either the KStars mailing list or bugs.kde.org)

At any rate, can you try using this?

```
bool CCDChip::capture(double exposure)
{
    switch (type)
    {
        case PRIMARY_CCD:
            clientManager->sendNewNumber(baseDevice->getDeviceName(), "CCD_EXPOSURE", "CCD_EXPOSURE_VALUE", exposure);
            break;

        case GUIDE_CCD:
            clientManager->sendNewNumber(baseDevice->getDeviceName(), "GUIDER_EXPOSURE", "GUIDER_EXPOSURE_VALUE", exposure);
            break;
    }

    return true;
}
```

k-ross, 2019-05-26:
I just tested your version, and it does not fix the problem. The version of sendNewNumber() that takes the list of explicit arguments just retrieves the INumberVectorProperty and updates the value, just like the original function. So, same result. 

I will file a new bug at bugs.kde.org (with patch), since this is the wrong place.

k-ross, 2019-05-26:
https://bugs.kde.org/show_bug.cgi?id=407952

closed 2019-06-05 by knro