indilib/indi #955: Configurable polling interval for SQM

skolarianer, 2019-05-29:
Please consider to make the polling interval of SQMs configurable. AFAIK the driver queries the SQM every second. So the connection to the unit is never really closed. There is no chance for other processes to access the SQM between readouts.

Our unit is connected to an SQM-network and reports measurements every 5 minutes. But when the indi driver is active no other requests are possible.

It'd be great if there was an option to change the polling interval and a closing of the socket after each readout.

knro, 2019-05-29:
It's already configuration in Options --> Polling. But you can set it to a max of 60000 ms now = 1 minute. I just increased the maximum limit to 600,000 or 10 minutes.

closed 2019-05-29 by knro

skolarianer, 2019-05-30:
Sorry, can't find it. Or do you mean other options?
In the options tab of the Pegasus PPB for example I can see polling settings.

[image]

knro, 2019-05-30:
Right it wasn't added. Please use the next nightly (v1.1 of the driver) and see if it has the control.

skolarianer, 2019-06-07:
The control element is now available.

It worked with a polling period of 1000ms. When I changed the period to 5000ms, it stopped working and I can’t set it back to 1000ms.

Can’t reconnect either. But when restarting the INDI services I can connect again, but do not get any data anymore. Sometimes I get this error in that case:

` [ERROR] Failed to parse input r, 00.00m,0000573909Hz,0000000000c,0000`

<details>
    <summary>Here is the full log:</summary>

```
[2019-06-07T15:12:02.524 CEST DEBG ][           org.kde.kstars.indi] - INDI: Stopping local drivers...
[2019-06-07T15:12:02.527 CEST DEBG ][           org.kde.kstars.indi] - Removing managed driver "CCD Simulator"
[2019-06-07T15:12:02.528 CEST DEBG ][           org.kde.kstars.indi] - INDIListener: Removing device CCD Simulator with unique label  ""
[2019-06-07T15:12:02.530 CEST DEBG ][           org.kde.kstars.indi] - Removing managed driver "SQM"
[2019-06-07T15:12:02.530 CEST DEBG ][           org.kde.kstars.indi] - INDIListener: Removing device SQM with unique label  "SQM"
[2019-06-07T15:12:02.533 CEST DEBG ][           org.kde.kstars.indi] - INDIListener: Removing client manager for server localhost @ 7624
[2019-06-07T15:12:02.534 CEST DEBG ][           org.kde.kstars.indi] - INDI server disconnected. Exit code: 0
[2019-06-07T15:12:02.535 CEST DEBG ][           org.kde.kstars.indi] - INDI server disconnected. Exit code: 0
[2019-06-07T15:12:02.535 CEST DEBG ][           org.kde.kstars.indi] - Stopping INDI Driver  "indi_simulator_ccd"
[2019-06-07T15:12:02.536 CEST DEBG ][           org.kde.kstars.indi] - Stopping INDI Driver  "indi_sqm_weather"
[2019-06-07T15:12:02.537 CEST DEBG ][           org.kde.kstars.indi] - Stopping INDI Server  "localhost" @ "7624"
[2019-06-07T15:12:02.538 CEST DEBG ][           org.kde.kstars.ekos] - Resetting Ekos Manager...
[2019-06-07T15:12:02.538 CEST WARN ][                       default] - QSqlDatabasePrivate::removeDatabase: connection 'filter_db' is still in use, all queries will cease to work.
[2019-06-07T15:12:02.539 CEST WARN ][                       default] - QSqlDatabasePrivate::addDatabase: duplicate connection name 'filter_db', old connection removed.
[2019-06-07T15:12:02.551 CEST INFO ][           org.kde.kstars.ekos] - "INDI services stopped."
[2019-06-07T15:12:09.311 CEST DEBG ][           org.kde.kstars.ekos] - Resetting Ekos Manager...
[2019-06-07T15:12:09.312 CEST WARN ][                       default] - QSqlDatabasePrivate::removeDatabase: connection 'filter_db' is still in use, all queries will cease to work.
[2019-06-07T15:12:09.312 CEST WARN ][                       default] - QSqlDatabasePrivate::addDatabase: duplicate connection name 'filter_db', old connection removed.
[2019-06-07T15:12:09.329 CEST INFO ][           org.kde.kstars.ekos] - "Starting INDI services..."
[2019-06-07T15:12:09.334 CEST DEBG ][           org.kde.kstars.indi] - INDI: Starting local drivers...
[2019-06-07T15:12:09.334 CEST DEBG ][           org.kde.kstars.indi] - Starting INDI Server:  ("-v", "-p", "7624", "-f", "/tmp/indififo5294f642") -f "/tmp/indififo5294f642"
[2019-06-07T15:12:09.340 CEST DEBG ][           org.kde.kstars.indi] - INDI Server Started?  true
[2019-06-07T15:12:09.340 CEST DEBG ][           org.kde.kstars.indi] - INDI: INDI Server started locally on port  7624
[2019-06-07T15:12:09.340 CEST DEBG ][           org.kde.kstars.indi] - Starting INDI Driver  "indi_simulator_ccd"
[2019-06-07T15:12:09.340 CEST DEBG ][           org.kde.kstars.indi] - Starting INDI Driver  "indi_sqm_weather"
[2019-06-07T15:12:09.340 CEST DEBG ][           org.kde.kstars.indi] - Adding managed driver "CCD Simulator"
[2019-06-07T15:12:09.340 CEST DEBG ][           org.kde.kstars.indi] - Adding managed driver "SQM"
[2019-06-07T15:12:09.341 CEST DEBG ][           org.kde.kstars.indi] - INDIListener: Adding a new client manager to INDI listener..
[2019-06-07T15:12:09.341 CEST DEBG ][           org.kde.kstars.indi] - INDI: Connecting to local INDI server on port  7624  ...
[2019-06-07T15:12:09.444 CEST DEBG ][           org.kde.kstars.indi] - INDI: Connecting to local INDI server on port  7624  ...
[2019-06-07T15:12:09.444 CEST DEBG ][           org.kde.kstars.indi] - INDI server connected.
[2019-06-07T15:12:09.444 CEST DEBG ][           org.kde.kstars.indi] - Connection to INDI server is successful
[2019-06-07T15:12:09.445 CEST INFO ][           org.kde.kstars.ekos] - "INDI services started on port 7624."
[2019-06-07T15:12:09.445 CEST DEBG ][           org.kde.kstars.indi] - Received new device SQM
[2019-06-07T15:12:09.473 CEST DEBG ][           org.kde.kstars.indi] - INDI Server:  "2019-06-07T13:12:09: startup: /usr/bin/indiserver -v -p 7624 -f /tmp/indififo5294f642 "
[2019-06-07T15:12:09.474 CEST DEBG ][           org.kde.kstars.indi] - INDI Server:  "2019-06-07T13:12:09: listening to port 7624 on fd 3"
[2019-06-07T15:12:09.475 CEST DEBG ][           org.kde.kstars.indi] - INDI Server:  "FIFO: start indi_simulator_ccd"
[2019-06-07T15:12:09.475 CEST DEBG ][           org.kde.kstars.indi] - INDI Server:  "FIFO: Starting driver indi_simulator_ccd"
[2019-06-07T15:12:09.476 CEST DEBG ][           org.kde.kstars.indi] - INDI Server:  "2019-06-07T13:12:09: Driver indi_simulator_ccd: pid=3665 rfd=4 wfd=7 efd=8"
[2019-06-07T15:12:09.476 CEST DEBG ][           org.kde.kstars.indi] - INDI Server:  "FIFO: start indi_sqm_weather -n \"SQM\""
[... cut ...]
[2019-06-07T15:12:19.435 CEST INFO ][           org.kde.kstars.ekos] - All INDI devices are now connected.
[2019-06-07T15:12:19.479 CEST DEBG ][           org.kde.kstars.indi] - < SQM >: < SKY_QUALITY >
[2019-06-07T15:12:19.489 CEST DEBG ][           org.kde.kstars.indi] - < SQM >: < Unit Info >
[2019-06-07T15:12:19.497 CEST INFO ][           org.kde.kstars.indi] - SQM :  "[INFO] Loading device configuration... "
[2019-06-07T15:12:19.524 CEST INFO ][           org.kde.kstars.indi] - SQM :  "[INFO] Device configuration applied. "
```
</details>

skolarianer, 2019-06-07:
Since the TCP connection can be interrupted by the XPort settings, `SQM::getReadings()` or `SQM::TimerHit()` should first check whether the TCP connection is still active and otherwise re-establish it.

In addition, the connection should be closed again after the data has been read in order to enable other processes to access the SQM, in cases where no hard disconnect is set in the XPort settings.

Maybe it’s a good idea to set the default period to at least 10000ms.

skolarianer, 2019-06-17:
Even if the "hard disconnect" is disabled in the XPort settings, the parsing error remains (most of the times).
I guess this is due to the concatenated serial number when the reporting feature is enabled in the SQM. (described in section 8.8 of the SQM-LE manual)

In this case the returned string looks like this:
r, 06.70m,0000022921Hz,0000000020c,0000000.000s, 039.4C,**00004217**

Here is the log …

```
2019-06-17T15:00:35: [ERROR] Failed to parse input c,0000000.000s, 048.3Cr, 00.00m,0000574400Hz,0000000000 
2019-06-17T15:00:05: [ERROR] Failed to parse input .000s, 048.3C,00004217r, 00.00m,0000574415Hz,0000000000 
2019-06-17T14:59:35: [ERROR] Failed to parse input .000s, 048.3Cr, 00.00m,0000574391Hz,0000000000c,0000000 
2019-06-17T14:59:05: [ERROR] Failed to parse input .000s, 048.0Cr, 00.00m,0000574390Hz,0000000000c,0000000 
2019-06-17T14:58:35: [ERROR] Failed to parse input 8.0C,00004217r, 00.00m,0000574371Hz,0000000000c,0000000 
2019-06-17T14:58:05: [ERROR] Failed to parse input 8.0Cr, 00.00m,0000574423Hz,0000000000c,0000000.000s, 04 
2019-06-17T14:57:35: [ERROR] Failed to parse input 8.0Cr, 00.00m,0000574416Hz,0000000000c,0000000.000s, 04 
2019-06-17T14:57:05: [ERROR] Failed to parse input 4217r, 00.00m,0000574356Hz,0000000000c,0000000.000s, 04 
2019-06-17T14:56:35: [ERROR] Failed to parse input 00.00m,0000574391Hz,0000000000c,0000000.000s, 047.7C,0000 
2019-06-17T14:56:05: [ERROR] Failed to parse input 00.00m,0000574411Hz,0000000000c,0000000.000s, 047.7Cr, 
2019-06-17T14:55:35: [ERROR] Failed to parse input 00.00m,0000574390Hz,0000000000c,0000000.000s, 047.7Cr, 
2019-06-17T14:55:05: [ERROR] Failed to parse input 00574384Hz,0000000000c,0000000.000s, 047.7C,00004217r, 
2019-06-17T14:54:35: [ERROR] Failed to parse input 00574425Hz,0000000000c,0000000.000s, 047.7Cr, 00.00m,00 
2019-06-17T14:54:05: [ERROR] Failed to parse input 00574439Hz,0000000000c,0000000.000s, 047.3Cr, 00.00m,00 
2019-06-17T14:53:35: [ERROR] Failed to parse input z,0000000000c,0000000.000s, 047.3C,00004217r, 00.00m,00 
2019-06-17T14:53:05: [ERROR] Failed to parse input z,0000000000c,0000000.000s, 047.3Cr, 00.00m,0000574393H 
2019-06-17T14:52:35: [ERROR] Failed to parse input z,0000000000c,0000000.000s, 047.3Cr, 00.00m,0000574472H 
2019-06-17T14:52:05: [ERROR] Failed to parse input 000c,0000000.000s, 047.0C,00004217r, 00.00m,0000574419H 
```

skolarianer, 2019-06-24:
Sorry for the late response. Unfortunately https://github.com/indilib/indi/commit/2e4e0a375056812d2b1ea53e184b91c79bede907 does not resolve the parsing issue completely. Because of a bug in the Unihedron firmware the serial no is sometimes also included in the "rx"-response.
If you use "Rx" instead the serial no is always included and can be simply discarded in this case.
See https://github.com/Noodlesalat/indi/commit/c04200f1ea49bb62d5ad7a88279571a0fcd02e02

@Noodlesalat will provide a PR.

The issue with the locked TCP connection still remains. Any ideas how to release the socket after each readout?

skolarianer, 2019-07-07:
I’m not sure this is going in the right direction. The point I’m trying to make regarding the TCP connection issue is described in section 6.1 "Writing your own software interface" of the SQM-LE manual:

> 4. Close the socket so that other programs can access the SQM-LE. Note: Only one connection can be made to the SQM-LE at a time. Therefore leaving a connection open constantly prevents other connections from being made.

The same applies for the interval reporting feature of the SQM. This feature opens a connection, sends the data and disconnects after the time specified in the web interface. If the TCP connection is blocked by INDI, no data can be sent in the meantime.

The goal is the following:
* The interval reporting feature opens a connection, sends the data and disconnects
* 2min pause
* INDI opens a connection, requests the data and disconnects/closes the TCP connection
* 2min pause
* start over

So IMHO this is not going to work:
`LOGF_WARN("Make sure SQM web timeout is configured for %ld seconds or more. Otherwise SQM will disconnect prematurely.", seconds);`
This is just preventing the disconnection of INDI’s connection, isn’t it? It must be the other way around. The disconnection time in the web interface must be shorter than the INDI polling intervall. And INDI must reestablish the TCP connection before every request and must disconnect afterwards. Otherwise the interval reporting feature and INDI block each other.

BTW: The default in the web interface is no disconnection at all. The disconnection is only manually enabled if you want to use the interval reporting feature as described in section "8.8.2 XPort settings for interval reporting mode" of the manual.

Disclaimer: I didn’t had the chance to test your commit yet.

commit b310d965 (2019-05-30): Add the polling control to the SQM driver. This should fix #955
commit 2e4e0a37 (2019-06-18): Flush before and after parsing readings. Should Fix #955
commit 885e2073 (2019-07-02): Ok this should hopefully fix #955

For longer polling period, you must configure SQM by the web interface first to increase the timeout (default 2 seconds).