indilib/indi #962: indibase/client build support for Android platform

laheller, 2019-06-11:
**Is your feature request related to a problem? Please describe.**
To make INDI clients for Android devices at least a client library for Android is needed. Therefore the build system for [libs/indibase](https://github.com/indilib/indi/tree/master/libindi/libs/indibase) can be extended also for Android.

**Describe the solution you'd like**
Create the necessary [Android.mk](https://developer.android.com/ndk/guides/android_mk) and [Application.mk](https://developer.android.com/ndk/guides/application_mk) files to be able to build libs/indibase for Android. Result would be the same like on Linux: _.so_ or _.a_ files.

**Describe alternatives you've considered**
N/A

**Additional context**
N/A

knro, 2019-06-11:
KStars Lite already runs on Android with the INDI Client library used here, but all the work to get it working on Android is done at KStars part. You want to make an explicit Android build for the client part?

laheller, 2019-06-11:
> KStars Lite already runs on Android with the INDI Client library used here, but all the work to get it working on Android is done at KStars part. You want to make an explicit Android build for the client part?

Hi @knro 
Yes, damn! KStars Lite! Of course it uses INDI client. :-)
Anyway yes, I would like to make an explicit Android build for the client.

BTW thanks for prompt reply!

Ladislav

knro, 2019-06-12:
Great, I have no idea how to do this, if you can submit a PR for this, it would be great!

laheller, 2019-06-13:
@knro 

I was able to build an Android _armeabi-v7a_ shared library from the following sources:
_baseclient.cpp basedevice.cpp indiproperty.cpp indistandardproperty.cpp ../indicom.c ../lilxml.c ../../base64.c_

But now I think it would be more easier (for end users of library) also to provide a **C wrapper** around _INDI::BaseClient_ class and provide static extern methods to access it.

knro, 2019-06-14:
Great progress. Ok, looking forward to the wrapper. Hopefully this would make Android development even more straight forward.

laheller, 2019-06-17:
@knro

Just a quick question:
To have a **minimal** INDI client (library) do we need to compile **all** the sources in [indibase](https://github.com/indilib/indi/tree/master/libindi/libs/indibase) directory or it's enough to have following ones:
_baseclient.cpp
basedevice.cpp
indiproperty.cpp
indistandardproperty.cpp
../indicom.c
../lilxml.c
../../base64.c_
?

BR

Ladislav

knro, 2019-06-17:
No need for all, check how the current client is built here: https://github.com/indilib/indi/blob/2e6051053dba0680d32674f577e94d037aaf0be1/libindi/CMakeLists.txt#L158

laheller, 2019-07-10:
@knro 
Android build is **working**, but I need some time to create the wrappers ( since I am coming from C# world, and have to learn a bit C/C++ :-) )

knro, 2019-07-17:
@laheller Great news! I'll await your PR.

knro, 2019-09-16:
Hello, is there any update on this?

laheller, 2019-09-20:
> Hello, is there any update on this?

Hello, I am still on it, but it will take a while, since I am learing the C++ (as I mentioned, coming from the C# and .NET world :) )

laheller, 2019-09-24:
Hi @knro 

I have some progress. I created a wrapper extension which basically wraps an implementation of INDI::BaseClient (like in the MyClient sample) and in the final library exposes "C" functions like _Initialize_ (new instance of MyClient), _Release_ (delete the reference) and all the inherited class methods. These functions will be then also available to high level languages (Java, Kotlin, C#).

Question: Is the INDI::BaseClient an universal class? I mean can be used to access whatever on the server side? I am just asking, because the example (tutorial_six) is only about camera device.

My goal is to make an universal Android native shared library which provides unrestricted access to whatever INDI device on the server side. Is the _INDI::BaseClient_ class enough to reach that?

BR,

Ladislav

knro, 2019-09-24:
INDI::BaseClient also uses INDI::BaseDevice.

You can see from the CMakeLists.txt that the client includes

basedevice.cpp
baseclient.cpp
indiproperty.cpp
indistandardproperty.cpp

knro, 2020-07-04:
Is there any update on this?

laheller, 2020-07-06:
> Is there any update on this?

Let's temporarily suspend or close this issue, since I have to work on another project....

knro, 2020-07-06:
Sure, please open this again when you have time! Good luck!

closed 2020-07-06 by knro