indilib/indi #963: Skywatcher Alt-Az mount doesn't work under Ubuntu 18.04

mpallewa, 2019-06-19:
Issue seen with ith the INDI Library v1.7.8 + Virtuoso mount. So I checked out the tree and debugged it.

The following hack resolves it,

diff --git a/libindi/drivers/telescope/skywatcherAPIMount.cpp b/libindi/drivers/telescope/skywatcherAPIMount.cpp
index f0d59734..3e9da621 100644
--- a/libindi/drivers/telescope/skywatcherAPIMount.cpp
+++ b/libindi/drivers/telescope/skywatcherAPIMount.cpp
@@ -37,9 +37,12 @@ namespace
 {
 bool FileExists(const std::string &name)
 {
+    return true;
+#if 0
     std::ifstream File(name.c_str());
 
     return File.good();
+#endif
 }
 } // namespace

knro, 2019-06-19:
@kecsap What do you think?

kecsap, 2019-06-19:
Huh, I can't say a definite opinion. I wanted to fix the connection issue in this summer, but I am on holiday now to verify the correctness. This driver does not connect since the serial driver refactoring. I myself reverted those changes in the spring to continue using the driver, but I was lazy to fix the real problem in the Skywatcher driver. If this fix works for the reporter you can merge this hack and I will look into the problem once back from holiday.

mpallewa, 2019-06-19:
I have tested with the virtuoso and the az-gti mount using an open drain serial cable (needed for virtuoso) with this driver and both seem to be working in preliminary testing.

knro, 2019-06-19:
@mpallewa Can you send in a PR? I know this is a simple change but a PR would be great (for any future changes as well).

mpallewa, 2019-06-19:
Done.

closed 2019-06-22 by knro