indilib/indi #982: Downscale of 16bit stream

nadvornik, 2019-07-06:
**Describe the bug**

ASI1600mm camera has 12bit ADC so the 16bit output has lowest 4 bits set to zero.
The downscale  code in StreamManager::asyncStream
https://github.com/indilib/indi/blob/master/libindi/libs/stream/streammanager.cpp#L326
takes lowest 8 bits so the result is effectively only 4 bit.

I think the downscale code should use ADC precision.

knro, 2019-07-07:
As you can see, the downscale code is just cutting off anything largest than 255 and is not performing proper 16bit to 8bit downscaling since this can be pretty slow. We need a multithreaded and smart downscaling that utilizes some of the modern processors capabilities on x86 and arm processors so that it won't causes a significant delay in streaming.

knro, 2019-07-07:
Ok, I'll test it with the zwo cameras I have and see if I can get better streaming.

nadvornik, 2019-07-08:
Right shift by 4 bits before cutting values over 255 should not be that slow.
The question is how to get the ADC precision of the camera.

knro, 2019-07-08:
I haven't tested ZWO yet.. does it report bit depth as 12bit in the main control panel?

nadvornik, 2019-07-08:
yes

ebertin, 2019-07-08:
That would be great! Note that in principle the raw data should be gamma-compensated before being streamed and displayed: checkout [this link](https://blog.johnnovak.net/2016/09/21/what-every-coder-should-know-about-gamma/) or (sorry for the self-citation) [this paper](http://adsabs.harvard.edu/abs/2012ASPC..461..263B).
Although realtime encoding at fast video rates may require a reasonably fast system with SIMD instructions (or a GPU).
Note that the KStars FITS viewer also lacks proper gamma compensation.

knro, 2019-07-09:
@ebertin Thanks for the reference. Many image related functions in both INDI & KStars could make use of SSE/SIMD instructions as now I'm just relying on the compiler to be smart enough about loops which isn't enough.

Do you know what's the best way to proceed for this downscaling to 8bit? You know about this a lot more than my superficial knowledge of the subject.

We have uint16_t input buffer, native bit depth (16, 14, 12, or 10) and we want uint8_t proper live stream of the data.

ebertin, 2019-07-09:
Well formally you should use equation (1):
output = round(255 * p)
where:
 p = 0 if I < 0
 p = a * I if 0 <= I <= Ii
 p = (1+b) * powf(I, 1.0/gamma) - b if Ii <= I <= 1
with:
I = input / maxrange (e.g., maxrange = 2^16-1)
and, for sRGB (OK for office use):
gamma = 2.4
a = 12.92
b  = 0.055
Ii = 0.00304
You can set other values, e.g. for HDTV (viewing in darker, higher contrast conditions).
The powf() is indeed a problem when it comes to speed. You may approximate the function above using a low-order polynomial, e.g., p = a0 + a1 * I + a2 * I^2 + a3 * I^3,
subject to the constraint that a0 = 0 and a1+ a2 + a3 = 1, and that the derivative matches that of the full formula at various points

knro, 2019-07-09:
@ebertin Thanks a lot, that would be useful. Looking at the current code, do you think you can submit a PR that we can test? You might be able to fine tune in the parameters for performance, quality..etc since you are an expect in this subject matter more than anyone here. It would greatly appreciated!

nadvornik, 2019-07-11:
I can look into it.

ebertin, 2019-07-11:
Yes sorry I am willing to help but I need to setup a development environment first and I won't have the time before the end of this month.

closed 2019-09-15 by knro