indilib/indi #983: No apparent way to get blob only using baseclient

pludov, 2019-07-10:
**Describe the bug**
I am trying to build a indi client that is supposed to capture only blob for one camera. 

However, BaseClient::listenINDI will always issue a <getProperties...> at start of the client. The side effect of this is unnecessary traffic being sent as well as some messages being re-emited by drivers that can't find their devices. (does it actually trigger the auto-detect ???)

My current workaround is to first watch a non existent device so that the <getProperties...> is actually a no-op:
``
    client->setServer("127.0.0.1",7624);
    client->watchDevice("Will never exists");
    client->connectServer();
    client->setBLOBMode(BLOBHandling::B_ONLY, "CCD Simulator", "CCD1");
``

However this seems very poor practice.

I propose to modify BaseClient to allow establishment of a true BLOB_ONLY connection:
  * BaseClient::setBLOBMode would work when called before connectServer (it currently try in that case, because it wants to emit the xml message over a not open fd)
  * BaseClient::connectServer would issue no getProperties when at least one blobMode has been set. It would instead issue pending blob_modes.

I am confident that this would not break existing clients since it is currently not possible to call setBLOBMode before connectServer, (leads to crash).
I can open a PR for that if it's fine for everyone

TallFurryMan, 2019-07-11:
Reading [the code](https://github.com/indilib/indi/blob/d0321cb7e11e582863f6025953a0f21f9a4a6fcb/libindi/libs/indibase/baseclient.cpp#L383), I understand that if you issued watchDevice with the device name before connecting the server, the induced ``getProperties`` should only pertain to that device name. I see no other part of BaseClient calling that... Do you have a log showing the effect?

Other than that, the idea sounds good to me. That would work the same as the watchDevice actually.

pludov, 2019-07-12:
You are right about watchDevice narrowing to one device. That's better than nothing.

Thinking twice, my previous solution would result in a client crash if using a non updated indilib, with no way for client code to detect that

i would rather add a new method to explicitely turn off the "general" getProperties when no watchdevice has been issued ( sth called setDefaultWatchDevice ).

TallFurryMan, 2019-07-12:
So I misread what you wanted to do: you would like to retrieve blobs from all devices once listening to the server? In your example you seem to target the CCD. I thought you would want to watch the CCD before connecting.

pludov, 2019-07-12:
I want to watch one device for blob only. So on the protocol side, no need for getproperties at all

TallFurryMan, 2019-07-13:
So if I understand correctly, your patch would be to test for the blob mode before issuing the getProperties call when you pre-set one or more devices to watch. That's one "if" at line 393 in baseClient.cpp testing the blob mode for each watched device. I don't think this would be a breaking change. Correct?

pludov, 2019-07-13:
This is my initial idea + plus make sure you can actually set the blob mode before connecting. (it currently crashes if you call it before connectServer)

But then, once corrected, if I write client code that rely on this new behavior, it'll crash wherever libindi is not updated to the latest rev (on the call to setBlobMode), and as a client writer there is no way to detect that neither at compile nor runtime. So that's pretty useless until libindi is updated everywhere

My second proposal is to include a new function to optionnaly disable the getProperties when no watchdevice has been issued. The first idea could still be implemented, at least to fix the crash on setBlobMode before connectServer

TallFurryMan, 2019-07-13:
OK understood (albeit slowly, sorry). My opinion is that you should not take precautions on the weaknesses of older indiservers. You found a bug, let's even call it a vulnerability as the server crashes, thus it is beneficial for everyone that you provide a fix. If the fix allows a subsequent PR from you that benefits to your intent, that's even better. No need to care about older indiservers. Crashes have to be fixed. @knro, your opinion?

pludov, 2019-07-13:
The problem is only on the client side.  The crash occur when trying to set blob mode without established connexion. The server is not impacted at all.

TallFurryMan, 2019-07-13:
Woops you're right, I'm not very focused today, sorry, thanks for your patience :) I see your problem now, there is no function returning the version of the installed INDI framework, only a #define at your build time that is mostly useless here. Solution is probably to put a requirement on the version of the lib on the system you install your client on, and let unsupported setups crash. Or manage the diversity with an option on your side :/

knro, 2019-07-14:
How about if we add watchProperties() function as well? So in watchDevice() we can also request which properties to watch for?

pludov, 2019-07-14:
Do you mean adding a "property" to the xml getproperties xml order ? That sure would be usefull but wouldn't it break protocol compatibilty ?

knro, 2019-07-14:
It's already implemented in indidriver.c (https://github.com/indilib/indi/blob/aba9f348b1a2e3004e9b31ae12cfc01e6ca682fa/libindi/indidriver.c#L930) for quite a while.

pludov, 2019-07-15:
Hello Jasem,
I tried to issue the following xml to indiserver, and it just consistently crashed the driver:
  <getProperties device="CCD Simulator" name="CCD1"/>

Probably driver adaptation would be required for this to work (my tests were done on the 1.7.8 tag)

knro, 2019-07-15:
Ok, I'll check this something like this can be implemented safely and with minimal impact to all.

commit cfc62975 (2019-07-18): add watchProperty to make client only request and only process properties of interest. Should fix #983 but it requires a lot of testing
closed 2019-07-18 by knro