indilib/indi #985: ghoto-ccd crash  since today update

joelcollet, 2019-07-17:
****Hello all,

Yesterday i used ekos  with gphoto d river with my olympus and, eenthough i had hard time to get  a capture  with it (still not solved), i was able to connect  without  issues to my olympus em5.
Today i performed an update of kstars and libini , together with the gphoto indi version.
gphoto is still working (able to connect and capture images) however the indi driver is crashing at onnection attempts
dmesg is full of  segfault like this one
indi_gphoto_ccd[7068]: segfault at fffffffffffffff9 ip 00007f0ccd3b3c74 sp 00007ffd502cdf10 error 5 in libc-2.27.so[7f0ccd31b000+1e7000]
[ 2025.765755] Code: c0 0f 85 8f 02 00 00 48 85 f6 48 89 f5 48 89 fb 0f 94 c0 48 85 ff 0f 95 c2 20 d0 0f 85 a5 02 00 00 48 85 ff 0f 84 4c 02 00 00 <48> 8b 4f f8 45 31 c0 4c 8d 67 f0 49 89 cf 49 83 e7 f8 f6 c1 02 0f 
[log_22-40-18.txt](https://github.com/indilib/indi/files/3403886/log_22-40-18.txt)
.
Hope this help
Regards
Joêl

joelcollet, 2019-07-17:
Sorry.looks that i got an issue with the update. After a clean reboot and a "new" update everything gets fine an running (except the liveiew not working, but was the case already before)
Joël

closed 2019-07-17 by joelcollet

knro, 2019-07-18:
This is with olympus over USB right? There was a request to add IP address connection that is not yet implemented.

joelcollet, 2019-07-18:
Hello Jasem 

Right, over usb. Now that i have done properly my upgrade, works fine. Nevertheless doing it by IP (over wifi then) would be really nice (removing cables spaghetti is always nice!!) 
I will ask to beryllius (on the indi forum) if he is able to get liveview or video..since i can't get it working. 

MAybe a useful link here for wifi control: 
https://github.com/LightSnowDev/OlympusWiFiControl 

REgards 
JoËl 

----- Mail original -----

joelcollet, 2019-07-18:
Hi Jasem 

i sent this mail to andrea at rebs dot biz : 
He worked on the OI share protocol (protocol used by olympus to remotely control their DSLR through wifi). This can help for the work on IP connection 

Hello Andreas 

I am currently the happy owner of an olympus EM5 mkII.. I ma using this DLSR for daytime phtography but also for astrophotography using a very nice software suite called kstars and ekos (indilib). We currently need to use the DSLR with an usb cable to get it working "remotely" or "tethered". Your tool is realy nice, since using http its is possible to do the very same, but without this cable (which is a bit "annoying" because of possible mess or spaghetti like entangle). It would be really nice to you if you can send us the command list you "discovered" and implemented in your software (or the source code of your software if you want) 
I will share this mail with jasem mutlaq which is the main dev of indilib software. 
Thanks for your work and nice tool 

Joël 

----- Mail original -----

joelcollet, 2019-07-18:
Hi Jasem 

Dunno if this can help : 
https://github.com/stv0g/libqt-omd 

https://gist.github.com/mangelajo/6fa005ff3544fecdecfa 

and joined is the command list of the OM-D em-5 mk II (got it through http://192.168.0.10/get_commandlist.cgi ) 

JoËl 

----- Mail original -----

joelcollet, 2019-07-18:
[get_commandlist.cgi.zip](https://github.com/indilib/indi/files/3408294/get_commandlist.cgi.zip)

knro, 2019-07-26:
the driver still needs to use libgphoto2 library commands to communicate with the camera. I think the above is for some server running on 3rd party software which we can't use.

joelcollet, 2019-07-26:
Yes, it is almost based on the eyeFi way of working (via get and post requests) which is different from connecting with gphoto via IP. 
Joël 

----- Mail original -----

totiherms, 2019-08-24:
Hi, I'm using a Canon EOS 4000D, and since I updated KStars to 3.3.4 the driver crashes after taking a picture. The log file:

[2019-08-24T20:01:35.831 CEST DEBG ][           org.kde.kstars.indi] - GPhoto CCD : "[DEBUG] Exposure complete. "
[2019-08-24T20:01:36.757 CEST DEBG ][           org.kde.kstars.indi] - GPhoto CCD : "[DEBUG] File added event completed. "
[2019-08-24T20:01:36.757 CEST DEBG ][           org.kde.kstars.indi] - GPhoto CCD : "[DEBUG] Downloading image... Name: (capt0000.jpg) Folder: (/) Delete from SD card? (false) "
[2019-08-24T20:01:36.757 CEST DEBG ][           org.kde.kstars.indi] - GPhoto CCD : "[DEBUG]  Downloaded 0x0 (preview 0x0) "
[2019-08-24T20:01:36.759 CEST DEBG ][           org.kde.kstars.indi] - INDI Server:  "2019-08-24T18:01:36: Driver indi_gphoto_ccd: capt0000.jpg: Not a TIFF or MDI file, bad magic number 55551 (0xd8ff)."
[2019-08-24T20:01:36.760 CEST DEBG ][           org.kde.kstars.indi] - INDI Server:  ""
[2019-08-24T20:01:36.761 CEST DEBG ][           org.kde.kstars.indi] - GPhoto CCD : "[DEBUG] Capture target is INTERNAL RAM. "
[2019-08-24T20:01:36.761 CEST DEBG ][           org.kde.kstars.indi] - INDI Server:  "2019-08-24T18:01:36: Driver indi_gphoto_ccd: Wrong JPEG library version: library is 62, caller expects 90"
[2019-08-24T20:01:36.762 CEST DEBG ][           org.kde.kstars.indi] - INDI Server:  ""
[2019-08-24T20:01:36.764 CEST DEBG ][           org.kde.kstars.indi] - INDI Server:  "Child process 601 died"
[2019-08-24T20:01:36.764 CEST DEBG ][           org.kde.kstars.indi] - Removing device GPhoto CCD
[2019-08-24T20:01:36.764 CEST DEBG ][           org.kde.kstars.indi] - INDI Server:  "2019-08-24T18:01:36: Driver indi_gphoto_ccd: stderr EOF"
[2019-08-24T20:01:36.764 CEST DEBG ][           org.kde.kstars.indi] - INDI Server:  "<delProperty device=\"GPhoto CCD\"/>"
[2019-08-24T20:01:36.764 CEST DEBG ][           org.kde.kstars.indi] - INDI Server:  ""
[2019-08-24T20:01:36.786 CEST DEBG ][           org.kde.kstars.indi] - INDIListener: Removing device GPhoto CCD with unique label  "GPhoto CCD"
[2019-08-24T20:01:36.787 CEST INFO ][     org.kde.kstars.ekos.align] - "World Coordinate System (WCS) is enabled. CCD rotation must be set either manually in the CCD driver or by solving an image before proceeding to capture any further images, otherwise the WCS information may be invalid."
[2019-08-24T20:01:36.813 CEST INFO ][           org.kde.kstars.ekos] - "GPhoto CCD is offline."
[2019-08-24T20:01:36.842 CEST INFO ][           org.kde.kstars.indi] - QHY CCD QHY5LII-M-6047e :  "[INFO] World Coordinate System is enabled. "
[2019-08-24T20:01:36.842 CEST DEBG ][           org.kde.kstars.indi] - < QHY CCD QHY5LII-M-6047e >: < CCD_ROTATION >
[2019-08-24T20:01:36.907 CEST DEBG ][           org.kde.kstars.indi] - INDI Server:  "2019-08-24T18:01:36: Driver indi_gphoto_ccd: Terminated after #0 restarts."
[2019-08-24T20:01:36.908 CEST DEBG ][           org.kde.kstars.indi] - INDI Server:  ""
[2019-08-24T20:01:36.908 CEST CRIT ][           org.kde.kstars.indi] - INDI driver  "indi_gphoto_ccd"  crashed!
[2019-08-24T20:01:39.380 CEST DEBG ][           org.kde.kstars.indi] - Stopping INDI Driver  "indi_gphoto_ccd"
[2019-08-24T20:01:39.381 CEST DEBG ][           org.kde.kstars.indi] - Removing managed driver "GPhoto CCD"

It works well with my previous KStars version, 3.2.3.

Any ideas?

thanks

totiherms, 2019-08-24:
I suppose that the crash is in the 

Driver indi_gphoto_ccd: Wrong JPEG library version: library is 62, caller expects 90

I'm running kstars on osx

knro, 2019-08-25:
@rlancaste Any idea about what libjpeg version is getting linked on MacOS? appears to be some mismatch.

rlancaste, 2019-08-25:
Ok I can look at that tomorrow

totiherms, 2019-08-27:
Hi, sorry but my camera just died... I wanted to try the new release, but when I connect the camera to USB it power off, seems a kind of short circuit...