indilib/indi #990: build for indi_ieqlegacy_telescope fails

fmozza, 2019-07-26:
**Describe the bug**
A clear and concise description of what the bug is.  Provide a general summary of the issue in the Title above as well as affected program name like `indiserver` or `indi_toupcam_ccd`

**To Reproduce**
Exact steps to reproduce the behavior.
1. Run driver
2. Perform action
3. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS:
 - Version [e.g. 1.7.6, GIT..etc]

**Log Files**
Make sure to enable logging and include log files https://indilib.org/support/logs-submission.html

fmozza, 2019-07-26:
indi_ieqlegacy_telescope build fails:  snippet from build log below:

Scanning dependencies of target indi_ieqlegacy_telescope
[ 81%] Building CXX object CMakeFiles/indi_ieqlegacy_telescope.dir/drivers/telescope/ieqprolegacydriver.cpp.o
/home/jmh/Devel/indi/libindi/drivers/telescope/ieqprolegacydriver.cpp:21:10: fatal error: ieqprolegacydriver.h: No such file or directory
 #include "ieqprolegacydriver.h"
          ^~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
CMakeFiles/indi_ieqlegacy_telescope.dir/build.make:62: recipe for target 'CMakeFiles/indi_ieqlegacy_telescope.dir/drivers/telescope/ieqprolegacydriver.cpp.o' failed
make[2]: *** [CMakeFiles/indi_ieqlegacy_telescope.dir/drivers/telescope/ieqprolegacydriver.cpp.o] Error 1
CMakeFiles/Makefile2:2921: recipe for target 'CMakeFiles/indi_ieqlegacy_telescope.dir/all' failed
make[1]: *** [CMakeFiles/indi_ieqlegacy_telescope.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2

system I'm building this on:
$ uname -a
Linux euler 4.4.132-1075-rockchip-ayufan-ga83beded8524 #1 SMP Thu Jul 26 08:22:22 UTC 2018 aarch64 aarch64 aarch64 GNU/Linux

Regards,

jmh

knro, 2019-07-27:
It's fixed now, thanks for the report.

closed 2019-07-27 by knro