indilib/indi #999: Slew command in LX200 Basic not issued properly

wbirk, 2019-08-18:
hello!

*Describe the bug**
after working on my telescope controller TSC for a while, I finally came back to solving an issue that bugged me for a while. I had/have the following problem.
- when connecting to the controller via LX200 Basic, all works fine
- when issuing a Slew command, the command is carried out but the position display in kstars is wrecked

after playing around in my code, I think i found the reason; "Slew" issues three commands, the desired position Sr and Sd, and a "Slew possible" command, :MS#.
Sr and Sd await a response, 1 if the coordinate is reasonable or 0 otherwise. I respond with "1" to each of these commands. Now here is the strange thing. INDI should
-send Sr ... and wait for the answer
-send Sd ... and wait for the answer
-send MS and wait for the answer

well, it does not; it sends Sd and MS as one command and apparently, does not expect two replies. as a result, the declination string and the RA string are read in the wrong order, and declination and RA are exchanged - which leads to a totally chaotic display, of course.

if i do not reply to :MS# as required in the LX200 protocol, everything is fine with my controller ...

**To Reproduce**
Check the commands sent by LX200 Basic when issuing a Slew command - Sr, Sd and MS. It appears that LX200 does not wait for the reply on the declination set. as a result, Declination and Right Ascension are switched IF your controller replies to all three commands as required in the LX200 protocol. there is simply one reply string too much ...

**Expected behavior**
LX200 Basic should handle the reply for all three commands

**Desktop (please complete the following information):**
 - OS: Ubuntu 18
 - Version: I do not have it hand, might be 1.6.8 ???
best regards wolfi

knro, 2019-08-19:
Got a log for that? There is no difference in GOTO between LX200 Basic and regular LX200 Telescope class as far as I can see.

wbirk, 2019-08-20:
hi!
will do a log as soon as possible ....
yours wolfi

closed 2020-02-24 by knro