indilib/indi #1000: [ERROR] Error starting exposure with Sony DSLR driver and Sony NEX-3N device

ccaramel, 2019-08-19:
I cannot make an exposure with Sony NEX-3N using Sony-dslr driver. 
Got every time the same error: [ERROR] Error starting exposure

I tried using gphoto CCD driver and I got the same behavior.

I'm on Osx Mojave with latest Kstars release 3.3.4 and its embedded INDI Library.
Tested using local INDI server.

I tested using a remote INDI server on RPI in order to test latest Nightly INDI Library and got exactly the same issue.

It worth mentioning that my Nex-3N is in bulb mode (Writed on its LCD screen).

Please find attached log file.

Is someone have an idea to overcome this issue?

Regards,

Christian.
[log_16-51-22.txt](https://github.com/indilib/indi/files/3516519/log_16-51-22.txt)

knro, 2019-08-20:
It gives a warning:

> Warning: Didn't find an exposure widget! Are you sure the camera is set to Bulb mode?

ccaramel, 2019-08-20:
I noticed the warning, but here it is a capture of the conf of my camera before connection USB
[image]

knro, 2019-08-20:
Thanks, then try this from the console and attach the gphoto.txt file.

`gphoto2 --debug --debug-logfile=gphoto.txt --capture-image-and-download`

ccaramel, 2019-08-20:
Thanks for your answer,
here it is:

[gphoto.txt](https://github.com/indilib/indi/files/3520282/gphoto.txt)

knro, 2019-08-21:
Ok it appears the MacOS version includes an older libgphoto2 library.

@rlancaste would it be possible to update to a more recent version?

rlancaste, 2019-08-21:
it would be possible to update it in the next release

ccaramel, 2019-08-21:
> Ok it appears the MacOS version includes an older libgphoto2 library.
> 
> @rlancaste would it be possible to update to a more recent version?

Hi, the gphoto2 debug command is made under the RPI (in order to let me use the latest library).
Do you think libgphoto2 library under RPI is an older one?
Here it is the return of gphoto2 -v on the RPI:
`gphoto2         2.5.15         gcc, popt(m), exif, cdk, aa, jpeg, readline
libgphoto2      2.5.23.1       all camlibs, gcc, ltdl, EXIF
libgphoto2_port 0.12.0         iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi, gcc, ltdl, USB, serial without locking
`

knro, 2019-08-21:
yeah it's old on the rpi3. Install the libgphoto2 ppa.

https://launchpad.net/~mutlaqja/+archive/ubuntu/libgphoto2

ccaramel, 2019-08-21:
I don't understand, libgphoto2 is already in version 2.5.23.1 wich seems to be the latest one
(cf. previous message)

rlancaste, 2019-08-21:
It is the latest as of June 2nd, but this is late August.  There have been 31 changes since then according to GitHub.  So yes, that version was the latest released version, but it is not the bleeding edge.

ccaramel, 2019-08-26:
Hi, using the latest release (August) I have exactly the same behavior.
It thus implies that sony_driver may not be updated in the next release?

ccaramel, 2019-08-27:
Hi, thank you for your answers.
The version I’m currently testing on Pi is the 4 weeks old ppa one.
And the bug reported in this thread is not solved, that’s why I think that it will not be solved in the next release.
Is there a way to test the bleeding edge version?

ccaramel, 2019-08-27:
Hi (again),
Tested beta of OSX Kstars, but connection to Sony NEX-3N is impossible.
Using both gphoto or sony-dslr driver.
Regards.

ccaramel, 2019-08-27:
Here it is.
[gphoto2.txt](https://github.com/indilib/indi/files/3546388/gphoto2.txt)

knro, 2019-08-27:
Ok try this again with the updated libgphoto2:

`gphoto2 --debug --debug-logfile=gphoto.txt --capture-image-and-download`

knro, 2019-08-27:
So even gphoto2 capture fails. It's probably a good idea to open a an issue in the libgphoto2 Github issue tracker and post the file above. Hopefully they can identify the issue and fix it?

ccaramel, 2019-08-27:
Done.
I inform you in any case.
Regards

rlancaste, 2019-09-09:
I compiled a new OS X version.  KStars 3.3.6.  I did find that there were some gphoto issues with the last build, this one resolved those issues.  You might try it to see if it works better.  Note that this won't resolve an issue if it is with gphoto itself, but should resolve a Mac OS only issue.

http://www.indilib.org/jdownloads/kstars/kstars-3.3.6.dmg

See if this helps.

knro, 2019-09-18:
Thanks for the report. Please open this issue in the 3rdparty repo issues --> https://github.com/indilib/indi-3rdparty/issues

closed 2019-09-18 by knro

pepcampas, 2020-01-05:
Have anybody found a solution for this?
Thanks