indilib/indi #1005: [indi-eqmod] Purging park position cause crash on next connetion

TallFurryMan, 2019-08-31:
**Describe the bug**
When the park position of `indi-eqmod` is purged, the content of ParkData.xml becomes `<parkdata/>`. This content crashes `indi-eqmod` on subsequent connects.

**To Reproduce**
Exact steps to reproduce the behavior.
1. Run `indiserver` with `indi-eqmod,` connect the `indi-eqmod` driver
2. Purge park data in the Site Management INDI panel
3. Disconnect the `indi-eqmod` driver.
4. Connect the `indi-eqmod` driver.

**Expected behavior**
Driver `indi-eqmod` connects and uses default park coordinates.

**Screenshots**
n/a

**Desktop (please complete the following information):**
 - OS: Gentoo kernel 4.4
 - Version `indi-eqmod` b07bf818

**Log Files**
not relevant, driver crashes while initializing

knro, 2019-08-31:
I believe this issue was fixed already... you're not using eqmod driver from HEAD?

TallFurryMan, 2019-08-31:
07bf81 is from May 16th. I just rebuilt the tip on the setup controller, I'll check the behavior with gdb.

knro, 2019-09-15:
Please reopen if you have this issue in master.

closed 2019-09-15 by knro

TallFurryMan, 2019-09-15:
Confirmed fixed on master.