indilib/indi #1008: [Atik] Driver does not detect device removal

TallFurryMan, 2019-09-07:
**Describe the bug**
After `indi-atik` is started and connected, unplugging the device has no effect on the driver. Subsequent commands just hang indefinitely. I have a bug in my `systemd` unit files which cause the driver to remain instead of stopping, but the driver should react to the unplug by failing its polling mechanism (set to 1000ms in the INDI panel).

**To Reproduce**
Exact steps to reproduce the behavior.
1. Plug device, start and connect driver, take a test exposure.
2. Unplug device, wait for a few minutes, the driver does not react.
3. Take a test exposure, the driver hangs indefinitely.

**Expected behavior**
When unplugging the device, after the polling interval, the driver should report an error and disconnect. It is up to the host system to restart or reconnect the driver when the device reappears.

**Screenshots**
n/a

**Desktop (please complete the following information):**
 - OS: Lubuntu 18.04 i386
 - Version [e.g. 1.7.6, GIT..etc]: 0bd6abc1

**Log Files**
n/a

knro, 2019-09-16:
Can the driver detect such a thing? it relies on the SDK for communication. The SDK should not hang and return with an error. So this needs to be reported to Atik directly.

closed 2019-09-16 by knro

TallFurryMan, 2019-09-16:
I see ignored libusb events popping up in the journal of the driver, but I did not delve into its implementation.

TallFurryMan, 2019-09-22:
Forwarded to Atik support, along with an additional question about frame dimensions.