indilib/indi #1009: indi_asi_ccd hangs on shutdown

docbobo, 2019-09-07:
**Describe the bug**
I've noticed that often times, there are multiple indi_asi_ccd process hanging around. While unclear to me if that is causing real issue, I am sure that's not exactly the desired behavior.

This problem is still happening with 2ba3f41e5a0b948d88ac7b327bb656f73c3f43b1, but I have seen it ever since starting with Indilib.

**To Reproduce**
No exact steps available.

**Expected behavior**
Only one indi_asi_ccd process running.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Debian Buster on RPi4
 - Version latest

**Log Files**
From gdb:

```
(gdb) bt
#0  futex_wait (private=0, expected=12, futex_word=0x723ab4) at ../sysdeps/unix/sysv/linux/futex-internal.h:61
#1  futex_wait_simple (private=0, expected=12, futex_word=0x723ab4) at ../sysdeps/nptl/futex-internal.h:135
#2  __pthread_cond_destroy (cond=0x723a90) at pthread_cond_destroy.c:54
#3  0xb684fc80 in std::condition_variable::~condition_variable() () from /usr/lib/arm-linux-gnueabihf/libstdc++.so.6
#4  0x004c8c30 in ASICCD::~ASICCD (this=0x71b2a0, __in_chrg=<optimized out>) at /home/pi/indi/3rdparty/indi-asi/asi_ccd.h:35
#5  0x004c8cbc in ASICCD::~ASICCD (this=0x71b2a0, __in_chrg=<optimized out>) at /home/pi/indi/3rdparty/indi-asi/asi_ccd.h:35
#6  0x004b6f64 in cleanup () at /home/pi/indi/3rdparty/indi-asi/asi_ccd.cpp:54
#7  0xb65e9670 in __run_exit_handlers (status=<optimized out>, listp=<optimized out>, run_list_atexit=run_list_atexit@entry=true, run_dtors=run_dtors@entry=true) at exit.c:108
#8  0xb65e9798 in __GI_exit (status=<optimized out>) at exit.c:139
#9  0xb6d0a08c in clientMsgCB (fd=0, arg=0x0) at /home/pi/indi/libindi/indidriver.c:873
#10 0xb6d10de4 in callCallback (rfdp=0xbe8d2bb4) at /home/pi/indi/libindi/eventloop.c:347
#11 0xb6d11290 in oneLoop () at /home/pi/indi/libindi/eventloop.c:439
#12 0xb6d0fe50 in eventLoop () at /home/pi/indi/libindi/eventloop.c:106
#13 0xb6d0fcd4 in main (ac=0, av=0xbe8d2dc4) at /home/pi/indi/libindi/indidrivermain.c:98
#14 0xb65d2718 in __libc_start_main (main=0xbe8d2dc4, argc=-1234153472, argv=0xb65d2718 <__libc_start_main+268>, init=<optimized out>, fini=0x5bebe0 <__libc_csu_fini>, rtld_fini=0xb6f2b4c4 <_dl_fini>, stack_end=0xbe8d2dc4)
    at libc-start.c:308
#15 0x004b6d24 in _start ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
(gdb)
```

docbobo, 2019-09-08:
I've updated to the latest version. Looks promising so far.

knro, 2019-09-15:
Please reopen if you encounter again.

closed 2019-09-15 by knro