indilib/indi #1010: Ekos capture module refocuses after any lucky frame.

Blueshawk, 2019-09-08:
Whenever I set an fwhm value to trigger a refocus the system reduces the value for any smaller frame, and then as soon as a slightly large frame comes in it triggers a refocus. With the focuser 
routinely moving way too far out and then getting stuck on "no stars" this causes the system to halt until wrangled manually back into shape. 

To reproduce: set a level for refocus and do a capture run. 
Error: Unit will alter your settings with each passing capture.
Expected behavior: It should not change that level if manually set. 

Linux Peppermint9 5.1.20-050120-generic x86_64 GNU/Linux
Ver. 3.3.4 2019-08-08T02:43:23Z
 
Okay...I know this was hashed out before when someone wanted it to be more reactive, but I think it's TOO reactive now, and is trying to do something that is impossible by continually trying to improve focus well beyond sky conditions. Maybe we could add a switch to tell it to hold the setting?
Thanks,
 Ray

knro, 2019-09-08:
Thank you for your report. This is a KStars issue, not INDI. Please file the bug at bugs.kde.org

closed 2019-09-08 by knro

Blueshawk, 2019-09-08:
aw dang. I'm on it.