indilib/indi #1014: Pulsar Dome Driver Installation

jmhirschi, 2019-09-16:
The Pulsar Dome Driver is not available on the site. The syntax to install is not correct. Can you please adapt the information on the related page.
Thank you.

knro, 2019-09-17:
Which syntax is incorrect?

jmhirschi, 2019-09-17:
Hello,
Thank you for your quick answer.
I followed the instructions below to add the Pulsar Driver to my Indi setup :
sudo apt-add-repository ppa:mutlaqja/ppa
sudo apt-get update && sudo apt-get install indi-bin
Unfortunately the driver is not available  when I try to add the Pulsar Dome under Ekos setup
----Message d'origine----

knro, 2019-09-17:
Right, it's available for INDI v1.8.2 (not released yet). It would be in the next release. Right now, you can build it from source or use the INDI Nightly PPA.

jmhirschi, 2019-09-17:
Thank you
I'll test it with the Nightly PPA

closed 2019-09-17 by jmhirschi

jmhirschi, 2019-09-19:
I've installed the Nightly PPA to test the dome. The driver is now visible in Ekos but when I launch the Indi I've an immediate crash of the Dome driver.