indilib/indi #1020: indiserver crashing with undefined symbol: _ZN4INDI13DefaultDevice11purgeConfigEv

mireianievas, 2019-09-24:
I've updated today my github in the Raspberry pi 3b+ (running raspbian) and adapted my build scripts to the now separated core/3rd party repositories. So far the drivers compile, but I get the folllowing when I run the indiserver

2019-09-24T11:50:40: Driver indi_simulator_ccd: indi_simulator_ccd: symbol lookup error: indi_simulator_ccd: undefined symbol: _ZN4INDI13DefaultDevice11purgeConfigEv
2019-09-24T11:50:40: Driver indi_simulator_ccd: stderr EOF
2019-09-24T11:50:40: Driver indi_simulator_ccd: restart #3

The same with the rest of the drivers.

knro, 2019-09-24:
You have an older INDI installed somewhere on the system. Check in /usr/local

You must make sudo make install after you clear any prior INDI installations.

mireianievas, 2019-09-24:
```
ls /usr/local/lib/ -l
total 58624
drwxr-sr-x 2 root staff     4096 Sep 24 09:49 arm-linux-gnueabihf
drwxr-sr-x 3 root staff     4096 Jun  7 22:31 indi
-rw-r--r-- 1 root staff   366146 Sep 24 13:14 libindiAlignmentClient.a
lrwxrwxrwx 1 root staff       27 Jun  7 22:31 libindiAlignmentDriver.so -> libindiAlignmentDriver.so.1
lrwxrwxrwx 1 root staff       31 Sep 12 09:50 libindiAlignmentDriver.so.1 -> libindiAlignmentDriver.so.1.8.1
-rw-r--r-- 1 root staff  1012464 Jun  7 21:38 libindiAlignmentDriver.so.1.7.8
-rw-r--r-- 1 root staff  1014260 Aug 24 03:34 libindiAlignmentDriver.so.1.8.0
-rw-r--r-- 1 root staff  1076516 Sep 24 13:13 libindiAlignmentDriver.so.1.8.1
-rw-r--r-- 1 root staff  1793380 Sep 24 13:12 libindiclient.a
-rw-r--r-- 1 root staff 14878026 Sep 24 13:15 libindidriver.a
lrwxrwxrwx 1 root staff       18 Jun  7 22:31 libindidriver.so -> libindidriver.so.1
lrwxrwxrwx 1 root staff       22 Sep 12 09:50 libindidriver.so.1 -> libindidriver.so.1.8.1
-rw-r--r-- 1 root staff  6301556 Jun  7 21:41 libindidriver.so.1.7.8
-rw-r--r-- 1 root staff  6351448 Aug 24 03:36 libindidriver.so.1.8.0
-rw-r--r-- 1 root staff  6673144 Sep 24 13:15 libindidriver.so.1.8.1
lrwxrwxrwx 1 root staff       17 Jun  7 22:31 libindilx200.so -> libindilx200.so.1
lrwxrwxrwx 1 root staff       21 Sep 12 09:50 libindilx200.so.1 -> libindilx200.so.1.8.1
-rw-r--r-- 1 root staff   244528 Jun  7 21:39 libindilx200.so.1.7.8
-rw-r--r-- 1 root staff   265172 Aug 24 03:34 libindilx200.so.1.8.0
-rw-r--r-- 1 root staff   272348 Sep 24 13:14 libindilx200.so.1.8.1
lrwxrwxrwx 1 root staff       14 Sep 24 13:22 libqhyccd.so -> libqhyccd.so.6
lrwxrwxrwx 1 root staff       18 Aug 24 03:44 libqhyccd.so.5 -> libqhyccd.so.5.1.0
-rw-r--r-- 1 root staff  6304720 Jun  7 22:37 libqhyccd.so.5.0.8
-rw-r--r-- 1 root staff  6322276 Aug 24 03:44 libqhyccd.so.5.1.0
lrwxrwxrwx 1 root staff       18 Sep 24 13:22 libqhyccd.so.6 -> libqhyccd.so.6.0.1
-rw-r--r-- 1 root staff  7103024 Sep 24 13:22 libqhyccd.so.6.0.1
drwxr-sr-x 2 root staff     4096 Sep 24 13:22 pkgconfig
drwxrwsr-x 4 root staff     4096 Sep 12 09:10 python2.7
drwxrwsr-x 3 root staff     4096 Jun  8 01:08 python3.7
```

mireianievas, 2019-09-24:
`indiserver -v
2019-09-24T13:43:04: startup: indiserver -v 
Usage: indiserver [options] driver [driver ...]
Purpose: server for local and remote INDI drivers
INDI Library: 1.8.1
Code v1.8.1-35-g0e15c513. Protocol 1.7.`

mireianievas, 2019-09-24:
Yes, I had an older installation of indi, I'm overwriting it (I'm already doing that sudo make install).

closed 2019-10-06 by knro

Blueshawk, 2022-02-11:
I realize this is old and closed but as it was the one I found that answered my similar problem I thought I'd add a note for others. In my case I got undefined symbol errors after installing only one new driver without updating the rest, not realizing that the difference in versions would cause this problem. 

Sure enough, when I updated the rest of indi* it solved this problem. 

Thanks @knro for all you do. :)