indilib/indi #1022: Double ZWO cameras issue

terrordrummer, 2019-10-02:
I own a ZWO ASI 178 mc and a ZWO ASI 120 mini cameras, I'm using the first as scope camera and the second as guide camera.
When Ekos starts the ZWO driver both cameras are recognized.
I'm able to start guiding with ZWO 120 mc while using the ZWO 178 for taking photos but very often camera acquisition times out and the whole session stops.
This happens in particular when I use the ZWO 128 mc as guiding camera, in average the image capturing exposure times out after few minutes, it's really unstable.

Not sure if this is a driver issue or ekos issue, when I use a single camera I can do everything without issues for the whole night.

When I'm using both camera exposure or guidance exposure times out very frequently.

Regards,
Roberto

knro, 2019-10-02:
Does the timeout happen when the primary camera is downloading?

knro, 2019-10-02:
also, please file ZWO issues at the [3rd party repository](https://github.com/indilib/indi-3rdparty) issues and not here.

closed 2019-10-02 by knro