indilib/indi #1025: MyFocuserPro2 indilib driver - the focuser "Max Position" value is not being read on focuser startup

CheekyMonkey, 2019-10-04:
**Describe the bug**
With the new MyFocuserPro2 indilib driver, the focuser "Max Position" value is not being read from the focuser controller on startup connection

**To Reproduce**
Steps to reproduce the issue are here : 
https://indilib.org/forum/focusers-filter-wheels/4802-myfocuserpro-or-myfocuserpro2-and-indi.html?start=48#44198

[video1](https://www.youtube.com/watch?v=SBz4k69xCEc)
[video2](https://www.youtube.com/watch?v=zo2e99d2SUA)

**Expected behavior**
The "Max Position" value should be read from the focuser controller on startup connection and used during focusing moves to prevent the focuser from travelling past the maximum focus position

**Desktop (please complete the following information):**
 - OS: Ubuntu Mate
 - Version: 18.04
 - Device: Raspberry Pi 4 (4Gb)

CheekyMonkey, 2019-10-06:
Update on testing Alan's new indi driver for MyFP2:

Following further tests and some debugging, it looks the issue with the new indi driver for MyFP2 is this:

In reading the focuser "Max Position", the indi driver uses the following which, of course, always returns zero:
(sendCommand(**":08#"**, res) == false)

I believe the code really should be:
if (sendCommand(**":8#"**, res) == false)

Found the issue whilst stepping/debugging the indi driver source:
https://www.youtube.com/watch?v=fG07z3VSWGY

I've corrected the issue in my local build of the driver 

I see that there's also a new MyFP2 firmware (v291) released in the last couple of days
I'm still on v290, but will flash v291 now as I see that there are provisions in that build regarding focus Max Position

Thanks Alan for building this new driver

knro, 2019-10-06:
Thanks, so is this 8 vs 08 corrected in version 291 of the firmware? or is that for something else?

@ajt68 can you please take a look?

CheekyMonkey, 2019-10-07:
Hi @knro and @ajt68 

The MyFP2 firmware is not a fault

The MyFP2 indi driver code should be corrected:

bool MyFocuserPro2::readMaxPos()
{
    char res[ML_RES] = {0};

    if (sendCommand(":08#", res) == false)
        return false;

**should become**

bool MyFocuserPro2::readMaxPos()
{
    char res[ML_RES] = {0};

    if (sendCommand(":8#", res) == false)
        return false;

commit 55ab4581 (2019-10-07): Fix MyFocuserPro2 max position readout on startup.

Fixes #1025
closed 2019-10-07 by knro