indilib/indi #1026: Coomunication errors between EKOS and LX200-compatible mount through TCP

fdesvallees, 2019-10-04:
I am testing communications through EKOS and a TeenAstro mount controller. This is derived from OnStep, but does not have a serial port. I communicate through TCP instead.
I used the LX200 Basic driver. The communication starts fine, I can read the initial coordinates of the mount, 

[2019-10-04T22:37:17.134 CEST INFO ][           org.kde.kstars.indi] - LX200 Basic :  "[INFO] Connecting to 192.168.0.12@9999 ... "
[2019-10-04T22:37:17.434 CEST DEBG ][           org.kde.kstars.indi] - LX200 Basic : "[DEBUG] Connection successful, attempting handshake... "
[2019-10-04T22:37:17.442 CEST DEBG ][           org.kde.kstars.indi] - LX200 Basic : "[SCOPE] CMD <:GR#> "
[2019-10-04T22:37:17.446 CEST DEBG ][           org.kde.kstars.indi] - LX200 Basic : "[SCOPE] RES <09:53:28> "
[2019-10-04T22:37:17.446 CEST DEBG ][           org.kde.kstars.indi] - LX200 Basic : "[SCOPE] VAL [9.89111] "
[2019-10-04T22:37:17.447 CEST INFO ][           org.kde.kstars.indi] - LX200 Basic :  "[INFO] LX200 Basic is online. "

but immediately after, the log shows communication errors

[2019-10-04T22:37:19.474 CEST DEBG ][           org.kde.kstars.indi] - LX200 Basic : "[SCOPE] CMD <:GR#> "
[2019-10-04T22:37:19.498 CEST DEBG ][           org.kde.kstars.indi] - LX200 Basic : "Error reading RA/DEC. "
[2019-10-04T22:37:21.484 CEST DEBG ][           org.kde.kstars.indi] - LX200 Basic : "[SCOPE] CMD <:GR#> "
[2019-10-04T22:37:21.491 CEST DEBG ][           org.kde.kstars.indi] - LX200 Basic : "Error reading RA/DEC. "
[2019-10-04T22:37:23.486 CEST DEBG ][           org.kde.kstars.indi] - LX200 Basic : "[SCOPE] CMD <:GR#> "
[2019-10-04T22:37:23.488 CEST DEBG ][           org.kde.kstars.indi] - LX200 Basic : "Error reading RA/DEC. "
[2019-10-04T22:37:25.488 CEST DEBG ][           org.kde.kstars.indi] - LX200 Basic : "[SCOPE] CMD <:GR#> "

I started investigating, and found out this is due to the tty_write() returning an error: first TTY_WRITE_ERROR then TTY_OVERFLOW.

Before continuing, I would like to ask if anyone recognizes this, and ask for suggestions on how to fix it. Thank you!

TallFurryMan, 2019-10-04:
I suppose you checked https://indilib.org/forum/development/5153-teenastro-driver-for-indi/38970.html ? You could probably get help by posting details there.

knro, 2019-10-06:
From the discussion at the topic above, it appears there needs to be a patch at the OnStep level.

fdesvallees, 2020-01-16:
Closed since TeenAstro firmware version 1.1, SHC version 1.2b which has an option in the Wifi menu to set a permanent IP connection ("One to One").

closed 2020-01-16 by fdesvallees