indilib/indi #1027: Problem with timers on device reconnect

nadvornik, 2019-10-05:
**Describe the bug**
I have noticed that the Moonlite Focuser increases the load on serial line with every reconnect, until the serial line is saturated and the focuser stops working.
The reason is this:

DefaultDevice::Connect() calls SetTimer on each connect.
https://github.com/indilib/indi/blob/master/libs/indibase/defaultdevice.cpp#L1026

MoonLite::TimerHit() keeps the timer running even if it is disconnected.
https://github.com/indilib/indi/blob/master/drivers/focuser/moonlite.cpp#L565

So with every reconnect, a new timer is added.

Fixing it in MoonLite::TimerHit()  with something like
 if (!isConnected()) return;
is not reliable if the disconnect-connect cycle is shorter than polling interval.

The same problem seems to be in other drivers, for example here:
https://github.com/indilib/indi/blob/master/drivers/focuser/deepskydad_af2.cpp#L915

I think that DefaultDevice::Connect() should check for already running timer.

**To Reproduce**
1. enable Driver debug in Moonlite focuser
2. do several disconnect-connect cycles
3. watch the frequency of debug messages increase

TallFurryMan, 2019-10-05:
Thanks for spotting this! Do you want to issue a PR with a proposal?

nadvornik, 2019-10-05:
Right now I am not sure what is the best fix, more specifically, how to tell if a timer id is still valid.
Maybe it needs a new function that finds the timer by the callback and user pointers.

knro, 2019-10-06:
The problem is in the MoonLite driver, it doesn't need to setTimer if it is disconnected.

commit a3cf040f (2019-10-06): Do not set timer if already disconnected.

Fixes #1027
closed 2019-10-06 by knro