indilib/indi #1029: RPi Tarbell file not available

TimColeAstro, 2019-10-07:
When clicking on the link for "Daily INDI Library for Raspberry Pi", the result is the following error message:
"The requested file was not found. Please inform the webmaster.: libindi_1.7.4_rpi.tar.gz"

garlick, 2019-10-13:
If this is where debs for raspbian pop out, well, that would indeed be nice to have!

knro, 2019-10-14:
we need someone from the community to step up and provides such debs (like launchpad built on daily/stable cycles).

garlick, 2019-10-14:
Is there a script or recipe available for building debs on raspbian? 

Hmm I wonder if it would be feasible to make a little raspbian build farm and run such a script against the github repo nightly, and upload results to the INDI download site?

tolson2000, 2019-10-14:
I am using Rasbian:
PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
VERSION_CODENAME=stretch

sudo apt-get install git

# Install INDI prerequisites
sudo apt-get install
libnova-dev libcfitsio-dev libusb-1.0-0-dev zlib1g-dev libgsl-dev 
build-essential cmake git libjpeg-dev libcurl4-gnutls-dev libtiff-dev 
libftdi-dev libgps-dev libraw-dev libdc1394-22-dev libgphoto2-dev 
libboost-dev libboost-regex-dev librtlsdr-dev liblimesuite-dev 
libftdi1-dev libfftw3-dev

sudo apt-get install gphoto2

Tom O.

garlick, 2019-10-14:
Thanks Tom - I was actually hoping for a script to build the indi debs direct from their github repos.

tolson2000, 2019-10-14:
Oh, OK, sorry. I don't bother with building debs; or rpms. Easy enough, 
with the prerequisites met, to git clone the repository; cmake; make; 
sudo make install. And I can customize.

garlick, 2019-10-14:
No worrles!  That's the current state of the practice for INDI on raspbian AFAIKT.

Ideally one could just  apt-get the INDI packages and pull in their dependencies automatically (and not their build dependencies).  It's just easier on users, and avoids various failure modes that come up over time, when packages are upgraded without tracking dependencies at the package level.

knro, 2019-12-27:
There is now Repo for it: https://indilib.org/download/raspberry-pi.html

closed 2019-12-27 by knro